mne-tools/mne-python #5273: ENH/IO start / stop arguments for EEG readers

dengemann, 2018-06-09:
Reading long EEG recordings can be a challenge on most computers. We should see which readers already support reading segments from raw and add start / stop args where ever needed. A current more urgent case is the EGI reader.

cc @agramfort

larsoner, 2018-06-09:
A workaround is to use `preload` as `str`, which will use memmapping (no restriction on memory size then)

dengemann, 2018-06-09:
That's actually true. It may just be slower. Looking at the more recently added .get_data in combination with preload I'm thinking we may actually be good already. Let me check out how it works in practice.

larsoner, 2018-06-09:
> That's actually true. It may just be slower. 

In theory it should be at least as fast as reading segment by segment from the disk directly (assuming the file and the memmapped file are on the same hard disk).

agramfort, 2018-06-10:
the recommended way is to do:

raw = read_raw...(preload=False)
raw.crop(tmin, tmax)
raw.load_data()

ideally this should work for all readers.

dengemann, 2018-06-10:
Good point. I was not remembering that crop works without preloading. I think we're good here. Perhaps we can add some content to FAQ.

dengemann, 2018-06-10:
Closing for now.

closed 2018-06-10 by dengemann