mne-tools/mne-python #5275: DOC: Explain core MNE concepts?

jona-sassenhagen, 2018-06-10:
We have a few terms that are MNE specific, or for which multiple names would make sense, but we've settled on one. We don't have a doc page going through all of them, right?

Would it make sense to have such a page?

Thinking of:
- selection
- group_by
- picks
- ...

sappelhoff, 2018-06-10:
+1 ... I was confused by "picks" in the beginning

agramfort, 2018-06-10:
+100

like sklearn glossary: http://scikit-learn.org/dev/glossary.html

can you start this?

cbrnr, 2018-06-11:
And even more basic things like raw, epoch, evoked, projection/projector, ...

agramfort, 2018-06-11:
ok I edited a bit the issue to start listing things we should put in the
glossary. Can you complement?

jona-sassenhagen, 2018-07-30:
I guess it should live in tutorials, and be referenced from the manual? I can make a sketch PR soon.

agramfort, 2018-07-30:
would be great

PR #5395 (2018-08-18): WIP: Glossary
A glossary of MNE terms.

 closes #5275
closed 2018-08-18 by larsoner