mne-tools/mne-python #5276: mne.preprocessing.ICA calculation of _pre_whitener

ckiefer0, 2018-06-11:
Line 538 in mne.preprocessing.ICA currently reads:
`pre_whitener[this_picks] = np.std(data[this_picks])[:, None]`

`np.std(data[this_picks])` calculates the standard deviation of all values of `data[this_picks]` taken together. As a result, all entries of `pre_whitener[this_picks]` are identical. However, based on my understanding of ICA, we should calculate the standard deviation for each channel separately.

The correct way to calculate the standard deviation on a per-channel basis should be this:
`pre_whitener[this_picks] = np.std(data[this_picks], axis=1)[:, None]`

cbrnr, 2018-06-12:
Good catch, thanks for spotting this. Fixed in #5277.

agramfort, 2018-06-12:
i was my decision originally. My intuition was that if we do as it is now,
PCA will more easily capture the blinks or ECG which lead to high amplitude
artifacts on a subset of sensors.

can you follow my reasoning?

cbrnr, 2018-06-12:
So this is basically the same as not scaling at all, right? I don't quite get it, because one assumption of PCA is that the input data should be equally scaled, otherwise it will always just find the components with the highest variance. For EEG, it shouldn't really make a huge difference to be honest, but scaling each channel individually makes more sense to me.

agramfort, 2018-06-12:
it's not equivalent if you have different channel types.

@pierreablin do you have an opinion?

cbrnr, 2018-06-12:
I see, this loop iterates over channel types. Yes, this works, but the cleaner (i.e. more correct) version is still scaling each channel individually within channel types IMO.

agramfort, 2018-06-12:
is it how eeglab or fieldtrip does it?

cbrnr, 2018-06-12:
Hm, EEGLAB zero-means all channels, but I don't see any scaling. We've discussed this in #4882, and there I called pre-whitening feature standardization, because I thought that each channel was scaled to unit variance. If you only scale within channel types, artifact channels retain a high variance, but overall different channel types have the same overall variance. I think this makes sense, but we might want to double-check with others. Also, this means that my updated comment is not accurate and we'd need to change it.

agramfort, 2018-06-12:
so I guess we can keep it as it is?

maybe @arnodelorme has an opinion?

pierreablin, 2018-06-12:
I don't really have an opinion on the question, but in the end if the ICA algorithm is not constrained to whiteness (e.g. infomax & picard) this won't make any difference.

cbrnr, 2018-06-13:
I think it is actually OK, but I will change the misleading comment. We don't want channels containing artifacts to be rescaled to the same variance as channels without artifacts, which would make it difficult for PCA to catch these artifacts. I agree with @pierreablin that for the subsequent ICA, it will not make a difference anyway.

cbrnr, 2018-06-13:
Actually the comment is OK, it says "Scale (z-score) the data by channel type".

cbrnr, 2018-06-13:
I'm closing this because scaling each channel individually would be detrimental (or at best wouldn't have an effect), see our discussion in #5276. If you think this is not how it should be done, please feel free to re-open @ckiefer0.

closed 2018-06-13 by cbrnr