mne-tools/mne-python #5279: Only permuting clusters that exceed min number of time-samples or contingent sources

LauraGwilliams, 2018-06-13:
Sorry if I've missed this in the documentation somewhere, but -- it seems like there is no way to run a permutation test (e.g. spatio_temporal_cluster_test, permutation_cluster_1samp_test) while first specifying the minimum spatial extent of the clusters to be tested (e.g. number of adjacent time samples or spatial sources).

Often, I think the user doesn't want to consider clusters that only last for one time sample, or only include a handful of sources. This has been a problem for me, because it means that the permutation test can take an extremely long time. Often it is permuting the results of thousands of tiny (short and small) clusters, that I know will be ultimately discarded.

Have I missed this -- is there a way of subsetting which clusters to perform the permutation test on?

If not, I could work on implementing it?

Thank you!

agramfort, 2018-06-13:
we could add this option for statistical control but I doubt this would
accelerate the
code as the bottleneck is in permuting data and computing connected clusters
that you will need to do anyway.

LauraGwilliams, 2018-06-13:
I see. I've seen implementations of this in the past that does the permutation for each found cluster, rather than permuting the whole dataset. I thought that's what was happening here.

I've implemented a simple thresholding that discards clusters that aren't of a certain size (combined time and spatial samples, for simplicity). Can I add a PR? You're totally right that it doesn't save much in the way of time, but it may be a useful addition.

agramfort, 2018-06-13:
> I see. I've seen implementations of this in the past that does the permutation for each found cluster, rather than permuting the whole dataset. I thought that's what was happening here.

I don't really see how this can work. You mean you remove some
features/tests in the permuted data?

> I've implemented a simple thresholding that discards clusters that aren't of a certain size (combined time and spatial samples, for simplicity). Can I add a PR? You're totally right that it doesn't save much in the way of time, but it may be a useful addition.

sure if you have it so we can see what it implies in terms of code change.

kingjr, 2018-06-13:
I think what you proposed is equivalent to smoothing / filtering the data.

LauraGwilliams, 2018-06-13:
> I don't really see how this can work. You mean you remove some features/tests in the permuted data?

If you find the adjacent sources/timepoints that exceed the statistical (say, t-value) threshold, you could then sum the t-values in the spatio-temporal cluster, to give the critical t-statistic. Then, permute the labels 10,000 times, and each time, sum the t-statistic over time and space just like in the critical test. Then, compare the observed t-sum to the distribution of t-sums.

I'm pretty sure this is the method I've seen in other places. Not sure if this would be faster? -- currently, the cluster tests seem extremely slow to me.

larsoner, 2019-08-21:
Closing since it sounds like there is nothing to change based on how we do the stats

closed 2019-08-21 by larsoner