mne-tools/mne-python #5280: Implement Topoplot Layout for 160-channel Yokogawa KIT MEG System

neurofractal, 2018-06-13:
A couple of people in our lab (https://www.cogsci.mq.edu.au/facilities/meg/) are starting to use MNE for BIDS data storage and analysis. We have a 160-channel Yokogawa KIT MEG System.

Data loads perfectly using mne.io.read_raw_kit. When trying to produce a topoplot after averaging I get the following error:

```matlab
"NotImplementedError: The layout for the KIT system with ID 345 is missing. Please contact the developers about adding it."
```

@teonbrooks or others, would this be easy to implement? Thanks in advance :) Robert

agramfort, 2018-06-13:
our layouts are there:

https://github.com/mne-tools/mne-python/tree/master/mne/channels/data/layouts

do you have one for your system? maybe from another software?

neurofractal, 2018-06-13:
We use:

```matlab
cfg             = [];
cfg.grad        = alldata.grad;
lay             = ft_prepare_layout(cfg, alldata);
```
... which creates a variable with position information.

In MNE layouts (e.g. KIT-157.lout) what do the columns represent? I'll have a go at creating one for our system.

PR #5314 (2018-07-03): Macquarie kit lay
taking over #5310

closes  #5280
closed 2018-07-03 by agramfort