mne-tools/mne-python #5281: Add code of conduct?

cbrnr, 2018-06-13:
Should we add a code of conduct to our project? Most projects seem to adopt the [Contributor Covenant](https://www.contributor-covenant.org/) these days. Thoughts? If we decide to adopt this as well, how can people report violations? I assume opening an issue is not the best way to do this, but I don't know if we have a project email address.

agramfort, 2018-06-13:
honestly I don't think MNE is big enough to necessitate this. So far I have
never
observed wrong behavior but maybe I am the one to blame? :)

larsoner, 2018-06-13:
SciPy recently adopted a contributor-covenant-like code of conduct, which includes how to report issues (they have members of a committee). I have no problem with us adding this if it seems worthwhile to pepole. So I'm +0.5 on it (not a full +1 because I also agree with @agramfort that our risk level is lower than other libs due to specificity of use cases)

cbrnr, 2018-06-14:
I agree with you that currently it is not really necessary. However, I think that having a code of conduct doesn't hurt either. Although MNE is much smaller than e.g. SciPy, it is also not a small project anymore. Since the Contributor Covenant obviously contains rules that I think we all support, we could just as well add it to the project so that we are as open and inclusive as possible. So also +0.5 from my side.

cbrnr, 2018-06-19:
Does anyone else have an opinion?

massich, 2018-06-25:
Actually, github uses those things for its metrics.
see this 
[image]

I'm +1 to pick up conde-of-conduct, issue template and PR template from a project similar to MNE-python, modify them, add them and point contribute.md to our `doc/contribute.rst` So that we get the bar to the right ;)

cbrnr, 2018-06-25:
Good point. I'm also +1 for getting all ticks, but I don't like issue and PR templates. They add lots of boilerplate that makes it difficult to get to the core of the issue/PR. Unless of course we can just write some recommendations how to open an issue/PR and we don't pre-fill the actual issue/PR text field.

massich, 2018-06-25:
I think that pre-fill issue/PRs a are good thing. I think that regular contributers would just ignore them, however I find it really helpful when I don't know a project.

As a scikit-learn core dev phrase it: 
> I like the issue/PR templates, I `ctrl+a` and start taping.

cbrnr, 2018-06-25:
I completely agree with this person. Adding this extra step is creating a lot of unnecessary work (even if it is just pressing Ctrl+a). I'd prefer to have an empty template, and for newbies we can still have some text that explains how to create good issues/PRs.

cbrnr, 2018-06-27:
@massich can you check if it is possible to use the existing `doc/contribute.rst` file? The [help page](https://help.github.com/articles/setting-guidelines-for-repository-contributors/) only talks about how to create a new file.

larsoner, 2018-06-27:
Can you look at what SciPy did? I think it was a google group with a few volunteer admins added

cbrnr, 2018-06-27:
Yes, they set up scipy-conduct@googlegroups.com (https://github.com/scipy/scipy/blob/master/doc/source/dev/conduct/code_of_conduct.rst).

Should we set up mne-conduct@googlegroups.com? Who will be on the committee (i.e. have access to this group)?

cbrnr, 2018-06-27:
As a more professional-looking alternative to an @googlegroups address (which would also be more appropriate from a privacy point of view), might it be possible to create an email account @martinos.org?

larsoner, 2018-06-27:
It might be possible but increases admin difficulty, so I'd rather not

larsoner, 2018-06-27:
As for access we'd need volunteers who are frequent contributors

cbrnr, 2018-06-27:
OK, then l'll go ahead and create the google groups address. Since I assume there won't be many violation reports, I volunteer to be on the committee. Who else? We need at least three people.

larsoner, 2018-06-27:
I can be on it, and also @agramfort unless he objects

agramfort, 2018-06-27:
you're in charge

cbrnr, 2018-07-09:
Now that we have our CoC, what about linking our existing contributing guide? Do you have time to look @massich?

Also, it would be great if we had short texts for our issue/PR templates. However, if GitHub insists on creating an actual template that users like me select and delete, it's probably not worth it. I don't think that this is possible, but maybe I didn't check hard enough. Anyone care to double-check?

agramfort, 2018-08-04:
we just miss an issue template now.

@cbrnr or @massich can you have a look?

agramfort, 2018-08-04:
see https://github.com/mne-tools/mne-python/community

cbrnr, 2018-08-06:
Have we decided that we need templates? It seems like adding this extra step just to turn all community features green is probably not worth it because most reports seem to be OK. But if we decide we need it, I'd go with minimal templates (one minimal PR template and two issue templates for bug reports and feature requests).

agramfort, 2018-08-06:
Issue template suggested to provide a test file and a code snippet to
replicate the error, as well as report mne.sys_info() would be nice.
It would save us some emails.

cbrnr, 2018-10-18:
Closing since we already added everything.

closed 2018-10-18 by cbrnr