mne-tools/mne-python #5286: "picks" attribute is null list in _sources_as_epochs/evoked

jshanna100, 2018-06-15:
"picks" attribute is null list in _sources_as_epochs/evoked in the case when add_channels is None. This breaks functionality when, for example, you want to pick individual components out of the resulting epochs/evoked object with .pick_channels, as this function require a "picks" array.

agramfort, 2018-06-15:
can you share a code snippet to expose / replicate the pb?

jshanna100, 2018-06-15:
In the else case below picks is an empty list.

def _sources_as_epochs(self, epochs, add_channels, concatenate):
        """Aux method."""
        out = epochs.copy()
        sources = self._transform_epochs(epochs, concatenate)
        if add_channels is not None:
            picks = [epochs.ch_names.index(k) for k in add_channels]
        else:
            picks = []
        out._data = np.concatenate([sources, epochs.get_data()[:, picks]],
axis=1) if len(picks) > 0 else sources

agramfort, 2018-06-17:
I am sorry I don't get the issue. This function is executed by a test and
it's fine
to do things like this:

>>> epochs.get_data()[:, []]
array([], shape=(1, 0, 241), dtype=float64)

jshanna100, 2018-06-17:
Yes, but it will fail if one calls e.g. .pick_channels on it, because that function expects an array to be in self.picks. One can hack around it by doing something like 

`epochs.picks = np.array(range(len(epochs.ch_names)))`

agramfort, 2018-06-17:
please share a full gist to replicate a crash so I don't have to reverse
engineer what you have in mind.

ideally paste it at: https://gist.github.com/

thanks

jshanna100, 2018-06-17:
I can no longer replicate the bug, and don't really have an explanation for it. Yesterday, the code required the hack I mentioned above in order to run, and today, it doesn't. Really sorry to waste your time...

closed 2018-06-17 by agramfort