mne-tools/mne-python #5292: Compensation channels pick error

Heneark, 2018-06-21:
While trying to execute `evoked.interpolate_bads()`, I ran into the following error:
```
  mne\channels\interpolation.py", line 190, in _interpolate_bads_meg
    info_to = pick_info(inst.info, picks_bad)

  mne\io\pick.py", line 411, in pick_info
    'channels' % (current_comp, comps_missing))

RuntimeError: Compensation grade 3 has been applied, but compensation channels are missing: ['BG1-2805', 'BG2-2805', 'BG3-2805', 'G11-2805', 'G12-2805', 'G13-2805', 'G22-2805', 'G23-2805', 'P11-2805', 'P13-2805', 'Q11-2805', 'Q13-2805', 'R11-2805', 'R12-2805', 'R13-2805', 'R22-2805', 'R23-2805']
Either remove compensation or pick compensation channels
```
This error seems to be raised automatically when attempting to make any pick that does not include all `ref_meg` channels. I dug a bit into compensation grades, and it seems that it can only be removed for Raw data using the `apply_gradient_compensation(0)` method.

Maybe it would be better to turn this error into a warning? Indeed in my case, compensation channels are not removed from data, the function only attempts to separately pick bad channels to compute interpolation.
Another way could be to always include these channels, similar to the `ref_meg='auto'` argument of `pick_types()`, but I'm not sure it would make sense in every case (especially this one).
Thanks,

larsoner, 2018-06-22:
Indeed this should probably not be an error, see our current plan for fixing these sorts of issues [here](https://github.com/mne-tools/mne-python/issues/5221#issuecomment-392057909).

Although for the case of `interpolate_bads`, I'm not sure what the right thing to do would be -- maybe we should go back to grade 0, interpolate, then re-apply whatever compensation was there originally. @bloyl @agramfort any ideas on this one?

Heneark, 2018-06-22:
I just tried to run my source reconstruction scripts, and it turns out this bug is everywhere (evoked.plot_joint, beamformer.make_lcmv), nothing works anymore since I upgraded from v0.15 to v0.16 last week.

larsoner, 2018-06-22:
Oof, that's not good. While we work on a proper fix, you could turn it into a `warnings.warn` in your local copy of the MNE code

agramfort, 2018-06-23:
could be related to https://github.com/mne-tools/mne-python/pull/5295

can you share a gist. We'll fix ASAP.

we need more MNE developers with CTF data...

agramfort, 2018-06-24:
@Heneark #5295 has been merged. Can you provide some test snippet so we can fix ASAP and release 0.16.2 

thanks

Heneark, 2018-06-25:
I downloaded the master branch, evoked.plot_joint() is indeed fixed.
Here's a test snippet for the interpolation error:
https://gist.github.com/Heneark/2e18280f88f956ae5faef3cea40e4bd0

Same error with beamformer:
```
filters = mne.beamformer.make_lcmv(evoked[k].info, fwd_surf, data_cov, noise_cov=noise_cov, pick_ori='vector')
Traceback (most recent call last):

  File "mne/beamformer/_lcmv.py", line 137, in make_lcmv
    _check_one_ch_type(info, picks, noise_cov)

  File "mne/beamformer/_compute_beamformer.py", line 119, in _check_one_ch_type
    info_pick = pick_info(info, sel=picks)

  File "mne/io/pick.py", line 408, in pick_info
    'channels' % (current_comp, comps_missing))

RuntimeError: Compensation grade 3 has been applied, but compensation channels are missing: ['BG1-2805', 'BG2-2805', 'BG3-2805', 'G11-2805', 'G12-2805', 'G13-2805', 'G22-2805', 'G23-2805', 'P11-2805', 'P13-2805', 'Q11-2805', 'Q13-2805', 'R11-2805', 'R12-2805', 'R13-2805', 'R22-2805', 'R23-2805']
Either remove compensation or pick compensation channels
```

Cheers,

agramfort, 2018-07-06:
have a look https://github.com/mne-tools/mne-python/pull/5321

it's not finished though.

closed 2018-07-09 by agramfort

agramfort, 2018-07-09:
@Heneark it should be fixed now. I also pushed a new version 0.16.2 to pypi.

you can do:

pip install -U mne

and hopefully everything should now work smoothly.

thanks for reporting the issue and your patience.

Heneark, 2018-07-11:
I upgraded to mne 0.16.2, `plot_joint()` is fixed, `interpolate_bads()` is fixed, but `pick_types(ref_meg=False)` still raises an error, thus the error from `beamformer.make_lcmv()` still occurs.

@agramfort Should I open another issue?

agramfort, 2018-07-11:
had you a opened a dedicated issue with this last issue with a code snippet
I can replicate?

larsoner, 2018-07-11:
I think you meant to link here

https://gist.github.com/Heneark/a183a24218302831860efc86c84c1950

Thanks for making it. We want to fix the problem generally, but the best solution will take some time. In the meantime it's quicker to squash the bugs one by one.

lclaudino, 2018-10-30:
Hello there

I'm having the same problem with the 0.16.2 version. I cannot `pick_types(ref_meg=False)` without gettting the same error as in the top post , and if I then `pick_types(ref_meg=True)`, `beamformer.make_lcmv` will raise the error instead. Could `beamformer.make_lcmv` be issuing the error because the number of channels in `info` and `forward` arguments are different (since `forward` will not include the compensation channels)? I don't want to change the compensation grade, so what I have been doing is to set `info['comps']=[]` and it works, but it is a hack.

Any ideas? Thanks!

agramfort, 2018-10-30:
you will need to wait for the next release or switch to using the current
development version.

See https://github.com/mne-tools/mne-python#get-the-latest-code

PR #5321 (2018-07-09): attempt to fix interpolation of bad MEG sensors with applied compensation
attempt to closes #5292

@larsoner @bloyl I need you help here.

to replicate the error run:

`pytest mne/channels/tests/test_interpolation.py -k test_interpolation_ctf_comp`

basically what happens is that doing `info_to = pick_info(inst.info, picks_bad)` here https://github.com/mne-tools/mne-python/blob/master/mne/channels/interpolation.py#L190

breaks as it wants to pick the info for just one data channel without the compensation channels.

help very much welcome.

cc @Heneark