mne-tools/mne-python #5297: MNE-Python installation error

massich, 2018-06-25:
This issue comes from MNE maling list ([here](https://mail.nmr.mgh.harvard.edu/pipermail//mne_analysis/2018-June/005033.html))

The problem can be replicated by doing this: 
```sh
❯ docker pull continuumio/miniconda
❯ docker run -i -t continuumio/miniconda /bin/bash
(base) root@b7f7fbaebb72:/# git clone --depth 1 https://github.com/mne-tools/mne-python.git
(base) root@b7f7fbaebb72:/# cd mne-python/
(base) root@b7f7fbaebb72:/mne-python# conda env create -f environment.yml

    ...
Building wheels for collected packages: mne, mayavi, nitime, nibabel, neo, pytest-sugar, sphinx-bootstrap-theme, python-picard, sphinx-fontawesome, pysurfer, sphinx-gallery, apptools, envisage, quantities, termcolor, configobj
  Running setup.py bdist_wheel for mne ... done
  Stored in directory: /root/.cache/pip/wheels/57/e7/02/814969527b7b2695368e12cd9aa25888819f268d01ab473886
  Running setup.py bdist_wheel for mayavi ... error
  Complete output from command /opt/conda/envs/mne/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-2uf7n_77/mayavi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-d_7cqh3z --python-tag cp36:
  running bdist_wheel
  running build
  Traceback (most recent call last):
    File "/opt/conda/envs/mne/lib/python3.6/site-packages/vtk/vtkRenderingOpenGL2.py", line 5, in <module>
      from .vtkRenderingOpenGL2Python import *
  ImportError: libGL.so.1: cannot open shared object file: No such file or directory
  
  During handling of the above exception, another exception occurred:
  
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/tmp/pip-install-2uf7n_77/mayavi/setup.py", line 500, in <module>
      **config
    File "/opt/conda/envs/mne/lib/python3.6/site-packages/numpy/distutils/core.py", line 169, in setup
      return old_setup(**new_attr)
    File "/opt/conda/envs/mne/lib/python3.6/site-packages/setuptools/__init__.py", line 129, in setup
      return distutils.core.setup(**attrs)
    File "/opt/conda/envs/mne/lib/python3.6/distutils/core.py", line 148, in setup
      dist.run_commands()
    File "/opt/conda/envs/mne/lib/python3.6/distutils/dist.py", line 955, in run_commands
      self.run_command(cmd)
    File "/opt/conda/envs/mne/lib/python3.6/distutils/dist.py", line 974, in run_command
      cmd_obj.run()
    File "/opt/conda/envs/mne/lib/python3.6/site-packages/wheel/bdist_wheel.py", line 202, in run
      self.run_command('build')
    File "/opt/conda/envs/mne/lib/python3.6/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/opt/conda/envs/mne/lib/python3.6/distutils/dist.py", line 974, in run_command
      cmd_obj.run()
    File "/tmp/pip-install-2uf7n_77/mayavi/setup.py", line 312, in run
      build_tvtk_classes_zip()
    File "/tmp/pip-install-2uf7n_77/mayavi/setup.py", line 298, in build_tvtk_classes_zip
      gen_tvtk_classes_zip()
    File "tvtk/setup.py", line 56, in gen_tvtk_classes_zip
      from tvtk.code_gen import TVTKGenerator
    File "/tmp/pip-install-2uf7n_77/mayavi/tvtk/code_gen.py", line 10, in <module>
      import vtk
    File "/opt/conda/envs/mne/lib/python3.6/site-packages/vtk/__init__.py", line 33, in <module>
      from .vtkRenderingOpenGL2 import *
    File "/opt/conda/envs/mne/lib/python3.6/site-packages/vtk/vtkRenderingOpenGL2.py", line 9, in <module>
      from vtkRenderingOpenGL2Python import *
  ModuleNotFoundError: No module named 'vtkRenderingOpenGL2Python'
  
...

CondaValueError: pip returned an error

(base) root@b7f7fbaebb72:/mne-python# 
```

<details>
 <summary>See the full log </summary>
```sh
~ 1m 6s
[... 377 lines cut ...]
        self.distribution.run_command(command)
      File "/opt/conda/envs/mne/lib/python3.6/distutils/dist.py", line 974, in run_command
        cmd_obj.run()
      File "/tmp/pip-install-2uf7n_77/mayavi/setup.py", line 312, in run
        build_tvtk_classes_zip()
      File "/tmp/pip-install-2uf7n_77/mayavi/setup.py", line 298, in build_tvtk_classes_zip
        gen_tvtk_classes_zip()
      File "tvtk/setup.py", line 56, in gen_tvtk_classes_zip
        from tvtk.code_gen import TVTKGenerator
      File "/tmp/pip-install-2uf7n_77/mayavi/tvtk/code_gen.py", line 10, in <module>
        import vtk
      File "/opt/conda/envs/mne/lib/python3.6/site-packages/vtk/__init__.py", line 12, in <module>
        from .vtkOpenGLKit import *
      File "/opt/conda/envs/mne/lib/python3.6/site-packages/vtk/vtkOpenGLKit.py", line 9, in <module>
        from vtkOpenGLKitPython import *
    ModuleNotFoundError: No module named 'vtkOpenGLKitPython'
   
    ----------------------------------------
Command "/opt/conda/envs/mne/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-2uf7n_77/mayavi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-71pgtdmk/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-install-2uf7n_77/mayavi/
 
CondaValueError: pip returned an error
 
(base) root@b7f7fbaebb72:/mne-python#
```
</details>

larsoner, 2018-06-25:
Their problem was missing gcc, the problem you created is missing OpenGL

massich, 2018-06-25:
@larsoner true. I saw `Failed building wheel for mayavi` and I didn't checked further

closed 2018-07-06 by agramfort