mne-tools/mne-python #5305: merge_grad_data and rescale in plot_tfr_topomap

Teekuningas, 2018-06-29:
Hi,

I was wondering if the order of merge_grad_data and rescale could be reversed in https://github.com/mne-tools/mne-python/blob/96bd8661599558024cdcd9edcc5eb0e12919722b/mne/viz/topomap.py#L1050

Now merging is losing the information of inhibition that is revealed by the rescale.

agramfort, 2018-06-29:
can you be more explicit? Not sure to follow what you mean by
"inhibition that is revealed by the rescale."

Teekuningas, 2018-06-29:
Or actually even better, use method='mean' in the merge_grad_data call. I don't see benefits in using 'rms' for power data. The problem is that when you use 'mag', 'planar1', 'planar2' or 'eeg' with baseline correction, you can get topomap that shows as red the areas where power has increased and as blue the areas where power has decreased compared to baseline. When you use 'grad', the plot will have only red colors as the merge_grad_data will force every value to be non-negative, and does that after rescaling.

larsoner, 2018-06-29:
@Teekuningas the reason `rms` is used rather than `mean` is that the grads are orthogonal in the tangential plane. So doing a plain `mean` will cause anything in the upper right or lower left of this tangential plane to add constructively, and anything in the lower right or upper left will cancel out. I can't think of a scenario where you would actually want this interpretation.

Teekuningas, 2018-06-29:
That probably makes sense; I kind of wanted a shortcut for also plotting a "difference tfr" (tfr1 - tfr2). But for a normal tfr, where every value is positive, it should be possible to first take the RMS and then rescale to have have meaningful plots where there is power increase and power decrease, shouldn't it?

agramfort, 2018-06-29:
I agree there should be a way to see ERD (blue) on grads.

the arithmetic could be done after combining grads for this.

PR + an example welcome

larsoner, 2018-06-29:
Agreed

Teekuningas, 2018-06-29:
Ok, I'll look into it!

closed 2020-04-11 by Teekuningas