mne-tools/mne-python #5309: possible bug in permutation_cluster_test

drammock, 2018-06-30:
I noticed that [here](https://github.com/mne-tools/mne-python/blob/master/mne/stats/cluster_level.py#L933) the PPF is only defined once for both `within` and `between` tests.  Shouldn't the `between` test be using `scipy.stats.f.ppf` instead of `scipy.stats.t.ppf` on [this line](https://github.com/mne-tools/mne-python/blob/master/mne/stats/cluster_level.py#L954)?  Also shouldn't the degrees of freedom be `dfn=(n_groups - 1), dfd=(total_subjs - n_groups)` instead of `*n_samples_per_group`?

agramfort, 2018-07-01:
@larsoner git blame suggest you wrote this:

https://github.com/mne-tools/mne-python/blame/master/mne/stats/cluster_level.py#L954

thoughts?

larsoner, 2018-07-01:
@agramfort that's just a refactoring commit, but indeed it introduced using `t.ppf` rather than `f.ppf`, which is a bug.

If you traverse the `git blame` to the line's origin, though, you find that the DOF was added by someone named @agramfort, so maybe you can ask him for some insight into that bit :)

https://github.com/mne-tools/mne-python/commit/6c31a6d175e941eed5a76cf291583d42ab24f97f#diff-23fe117cda04337c2d4a6c0fe30f0c76R804

I don't know offhand and would need to look into the math, which I can do in about a week.

agramfort, 2018-07-01:
looks like this guy @agramfort was correct in 2013 but his test was not
careful enough :)

larsoner, 2018-07-01:
I think @drammock is suggesting the DOF is incorrect, is it?

agramfort, 2018-07-01:
is there a test we could come up with?

drammock, 2018-07-02:
@agramfort I'm not sure what kind of test you mean...  it seems to me this is a simple question of whether the formula was correctly implemented.  Quoting from [wikipedia](https://en.wikipedia.org/wiki/F-test#Multiple-comparison_ANOVA_problems):

> This F-statistic follows the F-distribution with degrees of freedom d1=K−1 and d2=N−K under the null hypothesis [where K is the number of groups and N is the overall sample size].

A very simple example of how the current implementation is wrong is to give it more than 3 groups:
```
import numpy as np
from mne.stats import spatio_temporal_cluster_test
# 4 groups, 10 subjs/group, 50 timepoints, 1 vertex
fake_data = [np.random.randn(10, 50, 1) for _ in range(4)]
fvals, clusters, pvals, H0 = spatio_temporal_cluster_test(fake_data)
```
yields `TypeError: _parse_args() takes from 2 to 4 positional arguments but 5 were given` (because the `ppf` function is getting passed `*n_samples_per_group`, instead of just two DF args (`dfn` and `dfd`).

agramfort, 2018-07-02:
ok you made your point perfectly.

Any chance you can do the PR? at least start it?

thanks a lot

PR #5313 (2018-08-14): MRG: Fix permutation test
Closes #5309
closed 2018-08-14 by agramfort