mne-tools/mne-python #5311: KIT data doesn't retain MAG/GRAD type for channels

monkeyman192, 2018-07-02:
When importing KIT data we noticed that although the channel type is correct in the channel data (6001 for KIT data: FIFF.FIFFV_COIL_KIT_GRAD (we have gradiometers)), when observing the info of the raw we get `chs : list | 193 items (MAG: 125, MISC: 67, STIM: 1)`
Looking through the reason for this, it is because both MAG and GRAD type sensors are given `FIFF.FIFF_UNIT_T` as their unit, but when the channel type is determine by pick.py, the GRAD channel type has the `FIFF.FIFF_UNIT_T_M` type, so KIT data will never be realised to be a GRAD channel.
It is simple to add a few lines in the kit.py file to check whether the channel type is a gradiometer or magnetometer and set the units appropriately.
If this seems reasonable I can make a PR with the changes (unless there is a better way?)

agramfort, 2018-07-02:
@monkeyman192 : axial gradiometers like CTF with systems are considered by MNE as magnetometers on purpose. It's because they produce data that are much closer to a magnetometer as to a planer gradiometers in a vectorview system.

does it help?

agramfort, 2018-07-03:
agreed. Can you open an issue / PR in the mne-bids repo?

jasmainak, 2018-07-03:
yes, please let's discuss on mne-bids

larsoner, 2018-07-05:
Closing as non-MNE issue but feel free to reopen if there is something we need to do here

closed 2018-07-05 by larsoner