mne-tools/mne-python #5318: Conversion from Raw to ndarray

neeyanthkvk, 2018-07-05:
Hello,

I'm not sure if I am missing anything in the documentation, but many of the methods, including filter_data asks for the data in the format of a numpy array. Other methods, including the ICA methods ask for the data in the form of the Raw file. The ICA method in specific asks for the data to be high-pass filtered, but I cannot find an easy way to convert the Raw file to a numpy array (I know the reverse is possible with mne.io.RawArray)

cbrnr, 2018-07-05:
If you want to filter a raw object, you can use its `raw.filter` method, no need to convert to a NumPy array. If you need access to the NumPy array, you can access it via `raw._data`.

neeyanthkvk, 2018-07-05:
Is there a difference between RawEDF and the actual Raw classes? Most of the attributes seem to be the same (like raw.filter) but the raw._data doesn't seem to exist.

cbrnr, 2018-07-05:
Make sure you pass `preload=True` when loading the EDF file.

neeyanthkvk, 2018-07-05:
Yep! I see that now. Thanks!

jasmainak, 2018-07-05:
you should do `raw[:][0]` to get the numpy array. Don't use private attributes

closed 2018-07-05 by neeyanthkvk