mne-tools/mne-python #5320: raw.add_channels() creates duplicates in raw.info

ckiefer0, 2018-07-06:
I am using mne v0.15 for my anaIysis and I want to add a new stimulus channel to an existing raw file based on a list of events. However, raw.add_channels() seems to create duplicates in raw.info.

I wrote the following function which is based on the discussion [here:](https://github.com/mne-tools/mne-python/issues/4208)

    def add_events_to_stim_channel(raw, events, ch_name='ET_events'):

        picks = mne.pick_channels(raw.info['ch_names'], include=[ch_name])

        # if stim channel does not exist yet add new channel
        if picks.shape[0] == 0:
            info = mne.create_info([ch_name], raw.info['sfreq'], ['stim'])
            stim_data = np.zeros((1, raw.n_times))

            for evt in events:
                if evt is None:
                    print "No events found!"
                    return
                stim_data[0][evt[0]] = evt[2]

            stim_raw = mne.io.RawArray(stim_data, info)
            print 'stim_raw', stim_raw.info['bads']
            # add_channels adds duplicates to raw.info['bads'] and raw.info['dig']
            raw.add_channels([stim_raw], force_update_info=True)

        # if stim channel already exists add events to stim_channel
        else:
            raw.add_events(events, stim_channel=ch_name)

After using this function, all marked bad channels are listed twice in raw.info['bads'] and also all entries in raw.info['dig'] are present twice. I am not sure if any other entries are affected.

Is it possible that, for example, duplicates in raw.info['dig'] may affect future analysis?

agramfort, 2018-07-06:
can you provide a full gist to replicate and reveal the issue?

ckiefer0, 2018-07-06:
Here is the link to the [gist](https://gist.github.com/ckiefer0/b00fb36936dd1ec441eb7d7a2f30c7e3). The code as well as the raw file can also be found [here](https://www.dropbox.com/sh/1hh26djlqo7txpi/AAAXzsIpYshbGEJAwbtdHVeTa?dl=0).

agramfort, 2018-07-06:
the simplest way to replicate is this:

```
import os.path as op
import mne
from mne.io.meas_info import _merge_info

raw_fname = op.join(mne.datasets.sample.data_path(), 'MEG',
                    'sample', 'sample_audvis_raw.fif')

raw = mne.io.Raw(raw_fname, preload=True)

print(raw.info['bads'])
print(len(raw.info['dig']))

info = mne.create_info(["STI XXX"], raw.info['sfreq'], ['stim'])
info_out = _merge_info([raw.info, info], force_update_to_first=True)
print(info_out)
```

indeeds bads get duplicated. For the digs It does not seem so.

@larsoner do you have time to look?

PR #5327 (2018-07-10): MRG: Fix add_channels behavior
Closes #5320.

@ckiefer0 can you see if this fixes your problem?

@agramfort in the long run we should get rid of `info['buffer_size_sec']`, it is really a property of the Raw instance and not stored in `FIFFB_MEAS_INFO` block anyway.
closed 2018-07-10 by agramfort