mne-tools/mne-python #5324: Allow vlines=None in mne.viz.plot_compare_evokeds

sappelhoff, 2018-07-09:
I think we should allow for a `vlines='None'` argument in mne.viz.plot_compare_evokeds. Because in (arguably rare) some cases you would not want to have any vlines ...

It can be solved by a simple `if-clause` around the code here:

https://github.com/mne-tools/mne-python/blob/e4c995b4517089f7f105a9afb997f1b05a4e3653/mne/viz/evoked.py#L2087-L2096

we just wouldn't plot anything and pass `vlines=[]`to [`_setup_ax_spines`](https://github.com/mne-tools/mne-python/blob/55da4e49709ca6d5189cdceffba84c107b0ac517/mne/viz/utils.py#L2250)

And then also some additional `if-clause` some lines before that, when the input is being checked:

https://github.com/mne-tools/mne-python/blob/e4c995b4517089f7f105a9afb997f1b05a4e3653/mne/viz/evoked.py#L1859-L1861

This can just be skipped if we don't want vlines.

agramfort, 2018-07-09:
2 remarks:
- you can still remove these lines after the plot using the fig object
- if you argue it's too complicated no strong objection for a PR

jona-sassenhagen, 2018-07-30:
Can you pass an empty list already to `plot_compare_evokeds`? If not, maybe we should just make that possible by adding an if before the `ax.vlines`.