mne-tools/mne-python #5333: Support for reading brainvision exported from NeurOne Tesla amplifier

Kstoffer, 2018-07-11:
The Bittium NeurOne recording software is using some alternative spelling in their BrainVision export (.vhdr/.vmrk) files causing reading in MNE to fail, in order to enable reading of these I made some minimal changes to brainvision.py.
One could argue that the files should rather be fixed by Bittium - however reading via EEGLAB and fieldtrip work fine.
The changes are commented in the attached file, which I can push to git if desired.

Regards,
Kristoffer H. Madsen
Danish Research Centre for Magnetic Resonace
[brainvision.txt](https://github.com/mne-tools/mne-python/files/2185116/brainvision.txt)

larsoner, 2018-07-11:
Yes can you open a PR so we can properly view a diff?

Is it possible for you to make and export a very short (1 sec?) BrainVision file with the NeurOne export so we can add it to the `mne-testing-data` repository?

agramfort, 2018-07-11:
thx for reporting the issue.

any chance to you can send us a PR?

Kstoffer, 2018-07-11:
That was a crazy fast reply.
Yes, I will create a PR and also create a short file next time I'm in the
lab. I might not add the PR today as I want to check that it still works
with real Brainvision files first.

Regards,
Kristoffer

agramfort, 2018-07-11:
MNE is a pretty active community :)

larsoner, 2018-07-11:
> I might not add the PR today as I want to check that it still works with real Brainvision files first.

Our CIs will take care of that for you. We have lots of units tests including several BV variants, so if you open the PR you can see what breaks.

Locally you can test for likely failures with:
```
$ python -c "import mne; mne.datasets.testing.data_path(verbose=True)"
$ pytest mne/io/brainvision
$ make flake
```

agramfort, 2018-08-04:
@Kstoffer I took your patch that was coming for a rather old version of the master branch.

can you have a look at https://github.com/mne-tools/mne-python/pull/5392 and tell me if it works for you?

thanks

PR #5392 (2018-08-05): [MRG] fix for neuroone brain vision files
fixes #5333

cc @Kstoffer can you check if I did not forget anything?
closed 2018-08-05 by larsoner