mne-tools/mne-python #5335: Reading EGI Mff In Real Time

cascam07, 2018-07-12:
I am currently trying to read from an EGI Mff file in real time and not time locked to any events. Essentially I would like to read the last X seconds of data every X seconds and do some computation on that data. The simple (and bad) solution would be to just reload the file every time I want to read from it and grab the last X seconds of data. But this is obviously problematic as the Mff is getting continuously larger as the recording continues, thereby increasing the time it takes to load. For short intervals of X and long recording sessions, the time it takes to load the file will eventually exceed X, ruining the sampling interval.

Does MNE support a way to simply read the last X seconds of data from an Mff without reloading the whole thing?

Thank you very much,
Cameron

jasmainak, 2018-07-12:
If you use the fieldtrip buffer, you might be able to use [rt_client.get_data_as_epoch](https://martinos.org/mne/stable/generated/mne.realtime.FieldTripClient.html?highlight=get_data_as_epoch#mne.realtime.FieldTripClient.get_data_as_epoch). However, I don't know if this is possible for the EGI reader ...

larsoner, 2018-07-12:
> Does MNE support a way to simply read the last X seconds of data from an Mff without reloading the whole thing?

If you use `preload=False` it will only read in the header information when you do `read_raw_*`. Then to access the data you can do (I think?) `raw[:, -1000:]`

That is of course assuming the reader can properly ascertain the length of the file, which you'll have to try to find out

cascam07, 2018-07-13:
Thank you for the responses!

I tried the `raw[:, -1000:]` suggestion this morning but for some reason anything I tried using negative indexing returned an array of zeros for the data, even though the array of times was correct. To get around it I just did something like this:

`seglen = 15 #sample last 15 seconds of data`
`raw = read_raw_egi(egi_fname_mff, include=None, preload=False)`
`sfreq = int(raw.info['sfreq'])`
`nt = raw.n_times`
`data = raw[:, nt-(seglen*sfreq):nt-1]`

which seems to be working properly.

I did find that if I tried to read from the file more frequently than every 15 seconds, there would be instances where the data read would be the same chunk that was read previously. For example, if I were reading the file every 5 seconds, and the data I wanted was:
**A-B-C-D-E**..., 
with each letter representing a subsequent 5 second chunk of data. I would instead get:
**A-A-A-D-D**...

Fortunately 15 seconds works just fine for what I am trying to do, but I thought it might be worth mentioning.

Thank you again for your help!

larsoner, 2020-04-02:
Closing since it sounds like this was resolved. If there are remaining issues, [mne-realtime](https://github.com/mne-tools/mne-realtime) would probably be the right place to continue discussion

closed 2020-04-02 by larsoner