mne-tools/mne-python #5337: Output volume estimates as NIfTI-2?

jhouck, 2018-07-13:
Is it possible to output a volume STC as NIFTI-2 rather than NIFTI-1?  I have some fairly large files that exceed NIFTI-1's space constraints.  It looks like nibabel supports NIFTI-2 (of course it does), but I'm not sure how to get from a volume STC to a NIFTI-2 volume.   Would it be as simple as hacking _save_stc_as_volume to use nib.Nifti2Header and nib.Nifti2Image?

Thanks!

agramfort, 2018-07-14:
it's the line that needs to changed:

https://github.com/mne-tools/mne-python/blob/master/mne/source_estimate.py#L3191

jhouck, 2018-07-18:
Thanks, that did it (well, the analogous change in a slightly older dev version).  The files also open without error in programs that can read nifti-2.  Is this worth adding as an argument to the function?  It would be a week or so before I can get back to it.

larsoner, 2018-07-19:
Sure `format='nifti1'` default with new `'nifti2'` option would be good.

agramfort, 2018-07-19:
I would add a parameter. To be honest I don't have enough experience with
Nifti2 vs Nifti1 to make an educated decision.

you're in charge :)

jhouck, 2018-12-11:
Cleaning up my old open issues.  Looks like this was done as part of #5338 and https://github.com/mne-tools/mne-python/pull/5466, closing.

closed 2018-12-11 by jhouck