mne-tools/mne-python #5339: mne.Evoked.plot for different event types

carlosgmartin, 2018-07-13:
How can I get [mne.Evoked.plot](https://martinos.org/mne/dev/generated/mne.Evoked.html#mne.Evoked.plot) to plot the averages for different event types on the same axes, but in different colors?

agramfort, 2018-07-14:
can

https://martinos.org/mne/stable/generated/mne.viz.plot_compare_evokeds.html#mne.viz.plot_compare_evokeds

do the job?

This tutorial can be useful:

https://martinos.org/mne/stable/auto_tutorials/plot_visualize_evoked.html

carlosgmartin, 2018-07-14:
I'm getting `RuntimeError: No digitization points found.` when I try to set the `picks` parameter. Do you know why this might be happening?

jona-sassenhagen, 2018-07-30:
Can you show your code? It might be a bug.

jona-sassenhagen, 2018-09-23:
@carlosgmartin is this still an issue?

carlosgmartin, 2018-09-23:
I'm no longer having this issue.

closed 2018-09-23 by larsoner