mne-tools/mne-python #5342: BUG: CTF and proj (re)application

larsoner, 2018-07-16:
In principle we should be able to:

1. apply compensation or proj
2. compute forward on full set of channels
3. pick channels in data (maybe dropping compensation channels)
4. compute inverse without reapplying proj or compensation (#2310)

So we need to:

1. Always keep `info['comps']`.
2. Always keep `info['projs']` complete with the set of channels that were used when they were applied, i.e., do not subselect channels if `proj['active']`.
3. If projs are applied to raw, for `epochs(..., picks=[1], proj=True)`, raise an error telling people to use `proj=False` because they should not reapply?
4. Check for correspondence between `comps` and `proj` and data channels in the forward object during inverse application.

This non-reduction of the `projs`, however, would create some things to think about:
- In `master`, if you apply projs, then subselect channels, then compute the inverse, the projection operator is reapplied (I think?). This is a bit weird because it means that there were actually two spatial operators that got applied, but only the last (with fewer channels) gets applied to the lead fields.
- Currently we get the rank for inverse computation by subtracting the number of projectors, but this will only work if all channels are kept. We could still use `n_proj` -- we would just underestimate the rank, which is at least safer than overestimating it (and blowing up near-zero singular values).

This would probably take a lot of careful coding and testing to get right, but it should be doable. It seems like the cleanest option. @agramfort do you see any problems with this approach?

bloyl, 2018-09-16:
What do you think of breaking this up into multiple PRs?
with the first being to 

> Always keep info['comps']

and convert this RuntimeError to a RuntimeWarning
https://github.com/mne-tools/mne-python/blob/4f7f515d0dd3d035afb8eed58d313ab94304e9e8/mne/io/pick.py#L403-L408
This would revert us to pre #5133 status, and you/we would have slightly more time to properly triage this?

larsoner, 2018-09-16:
Seems reasonable as an intermediate step to me

larsoner, 2018-10-04:
We should do this intermediate step for 0.17 and the full fix for 0.18. @bloyl do you have time in the next two weeks to do the `RuntimeError`->`RuntimeWarning`? If not I can do it.

larsoner, 2018-10-04:
... and just a note that whatever fix we have should incorporate the gist that will eventually show up in #5384 :)

bloyl, 2018-10-15:
similar to #5610

if possible we should check that covariance objects match the compensation grade of data they are being applied to.

larsoner, 2018-10-16:
Unfortunately for `cov` IIRC we don't store the actual `info['chs']` field, just the `ch_names` so we can't check the grade

agramfort, 2018-10-16:
so are we screwed? what are your thoughts?

larsoner, 2018-10-16:
I'll have to look deeper at the code and the output. For example if we always store the uncompensated one, then it can be compensated then picked on the fly. It's possible this already happens, I haven't checked

dengemann, 2018-10-16:
@agramfort @larsoner greets from Lyon. People run into this issue and it blocks their progress in adopting MNE :) essentially what we see is that you have to drop the ref channels in order to apply refs. We see that excluding the reference channels is hard-coded that https://github.com/mne-tools/mne-python/blob/master/mne/proj.py#L74. I'm wondering what is the right thing to do here ...

dengemann, 2018-10-16:
We can see if we can repeat it on CTF sample data. What we see is that computing projs and applying them is broken if ref channels are present because the proj by default excludes them.