mne-tools/mne-python #5345: EDF read in not working in certain files, read in slow

raubsau, 2018-07-17:
1. 
The attached EDF file "edf_test.edf" can be opened via pyedflib:
```
Python 3.6.6 |Anaconda custom (64-bit)| (default, Jun 28 2018, 11:07:29) 
Type 'copyright', 'credits' or 'license' for more information
IPython 6.4.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import pyedflib

In [2]: f = pyedflib.EdfReader("edf_test.edf")

In [3]: f.readSignal(chn=0)
Out[3]: 
array([-18000.        , -18000.        , -18000.        , ...,
         4210.98649577,   4210.98649577,   4210.98649577])
```
File cannot be opened with mne:
```
In [1]: import mne

In [2]: mne.io.read_raw_edf("edf_test.edf", preload = True)
Extracting EDF parameters from edf_test.edf...
EDF file detected
Setting channel info structure...
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-2-8bfb048068a5> in <module>()
----> 1 mne.io.read_raw_edf("edf_test.edf", preload = True)

~/anaconda3/lib/python3.6/site-packages/mne/io/edf/edf.py in read_raw_edf(input_fname, montage, eog, misc, stim_channel, annot, annotmap, exclude, preload, verbose)
   1237     return RawEDF(input_fname=input_fname, montage=montage, eog=eog, misc=misc,
   1238                   stim_channel=stim_channel, annot=annot, annotmap=annotmap,
-> 1239                   exclude=exclude, preload=preload, verbose=verbose)

~/anaconda3/lib/python3.6/site-packages/mne/io/edf/edf.py in __init__(self, input_fname, montage, eog, misc, stim_channel, annot, annotmap, exclude, preload, verbose)

~/anaconda3/lib/python3.6/site-packages/mne/utils.py in verbose(function, *args, **kwargs)
    727         with use_log_level(verbose_level):
    728             return function(*args, **kwargs)
--> 729     return function(*args, **kwargs)
    730 
    731 

~/anaconda3/lib/python3.6/site-packages/mne/io/edf/edf.py in __init__(self, input_fname, montage, eog, misc, stim_channel, annot, annotmap, exclude, preload, verbose)
    156         input_fname = os.path.abspath(input_fname)
    157         info, edf_info = _get_info(input_fname, stim_channel, annot,
--> 158                                    annotmap, eog, misc, exclude, preload)
    159         logger.info('Creating raw.info structure...')
    160         _check_update_montage(info, montage)

~/anaconda3/lib/python3.6/site-packages/mne/io/edf/edf.py in _get_info(fname, stim_channel, annot, annotmap, eog, misc, exclude, preload)
    509         data_samps = np.delete(n_samps, slice(stim_channel, stim_channel + 1))
    510     sfreq = data_samps.max() * \
--> 511         edf_info['record_length'][1] / edf_info['record_length'][0]
    512 
    513     info = _empty_info(sfreq)

~/anaconda3/lib/python3.6/site-packages/numpy/core/_methods.py in _amax(a, axis, out, keepdims)
     24 # small reductions
     25 def _amax(a, axis=None, out=None, keepdims=False):
---> 26     return umr_maximum(a, axis, None, out, keepdims)
     27 
     28 def _amin(a, axis=None, out=None, keepdims=False):

ValueError: zero-size array to reduction operation maximum which has no identity
```
With nme, adding ```stim_chan = None``` solves the problem:
```
In [3]: mne.io.read_raw_edf("edf_test.edf", preload = True, stim_channel=None)
Extracting EDF parameters from edf_test.edf...
EDF file detected
Setting channel info structure...
Creating raw.info structure...In [1]: import mne

In [2]: mne.io.read_raw_edf("edf_test_edited.edf", preload=True)
Extracting EDF parameters from edf_test_edited.edf...
EDF file detected
EDF annotations detected (consider using raw.find_edf_events() to extract them)
Setting channel info structure...
Creating raw.info structure...
Reading 0 ... 1843199  =      0.000 ...  3599.998 secs...
Out[2]: <RawEDF  |  edf_test_edited.edf, n_channels x n_times : 2 x 1843200 (3600.0 sec), ~28.1 MB, data loaded>

Reading 0 ... 1843199  =      0.000 ...  3599.998 secs...
Out[3]: <RawEDF  |  edf_test.edf, n_channels x n_times : 1 x 1843200 (3600.0 sec), ~14.1 MB, data loaded>
```
The same EDF file with annotations (edf_test_edited.edf):
```
In [3]: mne.io.read_raw_edf("edf_test_edited.edf", preload = True)
Extracting EDF parameters from edf_test_edited.edf...
EDF file detected
EDF annotations detected (consider using raw.find_edf_events() to extract them)
Setting channel info structure...
Creating raw.info structure...
Reading 0 ... 1843199  =      0.000 ...  3599.998 secs...
Out[3]: <RawEDF  |  edf_test_edited.edf, n_channels x n_times : 2 x 1843200 (3600.0 sec), ~28.1 MB, data loaded>
```
Note that the EDF files only contain 1 channel, but ```<RawEDF  |  edf_test_edited.edf, n_channels x n_times : 2 x 1843200 (3600.0 sec), ~28.1 MB, data loaded>``` states ```n_channels = 2```.

2. 
Additionaly, EDF readin is really slow with ```stim_chan = None```:

Timing for read in of annotated file without ```stim_channel=None``` is really slow:
```
%timeit mne.io.read_raw_edf("edf_test_edited.edf", preload=True)
Extracting EDF parameters from edf_test_edited.edf...
EDF file detected
EDF annotations detected (consider using raw.find_edf_events() to extract them)
Setting channel info structure...
Creating raw.info structure...
Reading 0 ... 1843199  =      0.000 ...  3599.998 secs..
4.59 s ± 12.6 ms per loop (mean ± std. dev. of 7 runs, 1 loop each)
```

compared to read in with ```stim_chan=None```:
```
%timeit mne.io.read_raw_edf("edf_test_edited.edf", preload=True, stim_channel=None)
Extracting EDF parameters from edf_test_edited.edf...
EDF file detected
EDF annotations detected (consider using raw.find_edf_events() to extract them)
Setting channel info structure...
Creating raw.info structure...
<magic-timeit>:1: RuntimeWarning: 1 channel names are too long, have been truncated to 15 characters:
['EDF Annotations-1']
1 channel names are too long, have been truncated to 15 characters:
['EDF Annotations-1']
Reading 0 ... 1843199  =      0.000 ...  3599.998 secs...
59.2 ms ± 962 µs per loop (mean ± std. dev. of 7 runs, 10 loops each)
```

Can these issues be solved?

raubsau, 2018-07-18:
Thank you very much, I appreciate your answer.
Let me know if I can be of any help with testing etc.

cbrnr, 2018-07-18:
This might "just" be a documentation issue. Basically, you have an EDF file consisting of only one data channel (`edf_test.edf`). To load this file, you have to specify `stim_channel=None`, because otherwise it is assumed that the last (and only) channel is a stim channel. Of course this needs to be updated in the docs, or we can discuss whether we should change the default behavior to not parse stim channels in normal EDF files (without annotations).

The second test file (`edf_test_edited.edf`) contains two channels, and this time the default argument `stim_channel='auto'` correctly detects the stim channel because the file contains an annotation channel. Therefore, `stim_channel=None` does not make sense for this file, because this would mean that the stim channel is parsed as a regular data channel. I know this can be confusing, but MNE always creates a dedicated stim channel if it finds annotations. The important thing is that this channel is of type `STIM`, which is how you can distinguish it from other regular data channels.

Regarding the perceived sluggishness, I cannot reproduce this behavior. With the default `stim_channel='auto'`, it takes me around 140ms to load the file. It is true that the function is faster with `stim_channel=None` (around 42ms), but as I've said this does not result in the correctly parsed raw object. I've tried to vectorize [this potentially slow for-loop](https://github.com/mne-tools/mne-python/blob/master/mne/io/edf/edf.py#L371-L375), but I didn't get any noticeable speedup.

cbrnr, 2018-07-18:
FWIW, here's my vectorization attempt:

```
q, r = np.divmod(tal_channel_data.astype(np.int), 256)
tals = np.row_stack((r, q)).T.flatten().astype(np.uint8).tobytes()
```

The resulting `tal` byte string is identical to the bytes array object produces by the loop if there is only a single TAL channel. I have to think a bit more how to adapt this for the multi-channel case, but I didn't find a test for this case (let alone an example file).

@raubsau you could try to replace the lines I linked above with these two lines and see if it loads any faster.

raubsau, 2018-07-25:
@cbrnr here are the results after replacing the lines:

Without `stim_channel=None`:
```
In [1]: import mne

In [2]: %timeit mne.io.read_raw_edf("edf_test_edited.edf", preload=True)
Extracting EDF parameters from edf_test_edited.edf...
EDF file detected
EDF annotations detected (consider using raw.find_edf_events() to extract them)
Setting channel info structure...
Creating raw.info structure...
Reading 0 ... 1843199  =      0.000 ...  3599.998 secs...
214 ms ± 4.66 ms per loop (mean ± std. dev. of 7 runs, 1 loop each)
```

With `stim_channel=None`
```
In [1]: import mne

In [2]: %timeit mne.io.read_raw_edf("edf_test_edited.edf", preload=True, stim_channel=None)
Extracting EDF parameters from edf_test_edited.edf...
EDF file detected
EDF annotations detected (consider using raw.find_edf_events() to extract them)
Setting channel info structure...
Creating raw.info structure...
<magic-timeit>:1: RuntimeWarning: 1 channel names are too long, have been truncated to 15 characters:
['EDF Annotations-1']
1 channel names are too long, have been truncated to 15 characters:
['EDF Annotations-1']
Reading 0 ... 1843199  =      0.000 ...  3599.998 secs...
58.4 ms ± 689 µs per loop (mean ± std. dev. of 7 runs, 10 loops each)
```

cbrnr, 2018-07-25:
So this would be a 21x speedup from 4.59s to 0.214s. Just to make sure, could you run the old code with `stim_channel=None` again to see if it really takes that long?

cbrnr, 2018-07-25:
Sorry, I meant *without* `stim_channel=None`.

raubsau, 2018-07-25:
**mne 0.16.2, patched:**
```
import mne
%timeit mne.io.read_raw_edf("edf_test_edited.edf", preload=True)
Extracting EDF parameters from edf_test_edited.edf...
EDF file detected
EDF annotations detected (consider using raw.find_edf_events() to extract them)
Setting channel info structure...
Creating raw.info structure...
Reading 0 ... 1843199  =      0.000 ...  3599.998 secs...
253 ms ± 26.9 ms per loop (mean ± std. dev. of 7 runs, 1 loop each)

len(raw.find_edf_events())
Out[6]: 1

import mne
%timeit mne.io.read_raw_edf("edf_test_edited.edf", preload=True, stim_channel=None)
Extracting EDF parameters from edf_test_edited.edf...
EDF file detected
EDF annotations detected (consider using raw.find_edf_events() to extract them)
Setting channel info structure...
Creating raw.info structure...
<magic-timeit>:1: RuntimeWarning: 1 channel names are too long, have been truncated to 15 characters:
['EDF Annotations-1']
1 channel names are too long, have been truncated to 15 characters:
['EDF Annotations-1']
Reading 0 ... 1843199  =      0.000 ...  3599.998 secs...
67.9 ms ± 1.82 ms per loop (mean ± std. dev. of 7 runs, 10 loops each)
len(raw.find_edf_events())
Out[5]: 0
```

Testing with an EDF file of 24h length and 168 annotations (unfortunately I cannot share it):
```
import mne
%timeit mne.io.read_raw_edf(input_fname="bigfile.edf", preload=True)
Extracting EDF parameters from bigfile.edf...
EDF file detected
EDF annotations detected (consider using raw.find_edf_events() to extract them)
Setting channel info structure...
Creating raw.info structure...
Reading 0 ... 44236799  =      0.000 ... 86399.998 secs...
12 s ± 3.47 s per loop (mean ± std. dev. of 7 runs, 1 loop each)
len(raw.find_edf_events())
Out[4]: 168

import mne
%timeit mne.io.read_raw_edf(input_fname="bigfile.edf", preload=True, stim_channel=None)
Extracting EDF parameters from bigfile.edf...
EDF file detected
EDF annotations detected (consider using raw.find_edf_events() to extract them)
Setting channel info structure...
Creating raw.info structure...
<magic-timeit>:1: RuntimeWarning: 1 channel names are too long, have been truncated to 15 characters:
['EDF Annotations-1']
1 channel names are too long, have been truncated to 15 characters:
['EDF Annotations-1']
Reading 0 ... 44236799  =      0.000 ... 86399.998 secs...
2.03 s ± 113 ms per loop (mean ± std. dev. of 7 runs, 1 loop each)
len(raw.find_edf_events())
Out[4]: 0
```

**mne 0.16.2, unpatched:**
```
import mne
%timeit mne.io.read_raw_edf("edf_test_edited.edf", preload=True)
Extracting EDF parameters from edf_test_edited.edf...
EDF file detected
EDF annotations detected (consider using raw.find_edf_events() to extract them)
Setting channel info structure...
Creating raw.info structure...
Reading 0 ... 1843199  =      0.000 ...  3599.998 secs...
4.35 s ± 107 ms per loop (mean ± std. dev. of 7 runs, 1 loop each)
len(raw.find_edf_events())
Out[4]: 1

import mne
%timeit mne.io.read_raw_edf("edf_test_edited.edf", preload=True, stim_channel=None)
Extracting EDF parameters from edf_test_edited.edf...
EDF file detected
EDF annotations detected (consider using raw.find_edf_events() to extract them)
Setting channel info structure...
Creating raw.info structure...
<magic-timeit>:1: RuntimeWarning: 1 channel names are too long, have been truncated to 15 characters:
['EDF Annotations-1']
1 channel names are too long, have been truncated to 15 characters:
['EDF Annotations-1']
Reading 0 ... 1843199  =      0.000 ...  3599.998 secs...
60.7 ms ± 4.77 ms per loop (mean ± std. dev. of 7 runs, 10 loops each)
len(raw.find_edf_events())
Out[4]: 0
```

Testing with an EDF file of 24h length and 168 annotations:
```
import mne
%timeit mne.io.read_raw_edf(input_fname="bigfile.edf", preload=True)
Extracting EDF parameters from bigfile.edf...
EDF file detected
EDF annotations detected (consider using raw.find_edf_events() to extract them)
Setting channel info structure...
Creating raw.info structure...
Reading 0 ... 44236799  =      0.000 ... 86399.998 secs...
Extracting EDF parameters from bigfile.edf...
EDF file detected
EDF annotations detected (consider using raw.find_edf_events() to extract them)
Setting channel info structure...
Creating raw.info structure...
Reading 0 ... 44236799  =      0.000 ... 86399.998 secs...
1min 52s ± 3.4 s per loop (mean ± std. dev. of 7 runs, 1 loop each)
len(raw.find_edf_events())
Out[4]: 168

import mne
%timeit  mne.io.read_raw_edf(input_fname="bigfile.edf", preload=True, stim_channel=None)
Extracting EDF parameters from bigfile.edf...
EDF file detected
EDF annotations detected (consider using raw.find_edf_events() to extract them)
Setting channel info structure...
Creating raw.info structure...
<magic-timeit>:1: RuntimeWarning: 1 channel names are too long, have been truncated to 15 characters:
['EDF Annotations-1']
1 channel names are too long, have been truncated to 15 characters:
['EDF Annotations-1']
Reading 0 ... 44236799  =      0.000 ... 86399.998 secs...
1.81 s ± 23.5 ms per loop (mean ± std. dev. of 7 runs, 1 loop each)
len(raw.find_edf_events())
Out[4]: 0
```

Conclusions:
patching speeds up the reading in;
with`stim_channel=None` annotations cannot be accessed.

cbrnr, 2018-07-26:
Thanks @raubsau, this is valuable information. So we should definitely try to replace the existing slow loop with vectorized code. I need to spend some time tweaking the code, because currently it only works for a single stim channel. However, EDF+ supports more than one stim channel, and I need to take this into account.

And yes, if `stim_channel=None` it is expected that you cannot access annotations, because this is telling the function that there are no annotations (= no stim channels) in the data file.

Regarding the default, I need input from others. Should the default of `stim_channel='auto'` continue to use annotations if they are present (EDF+) and if not fall back to take the last channel as stim channel (EDF)? Or should we change the fallback to no stim channel? I'd vote for the former, because most EDF/BDF files I've seen (including the ones from our lab) do not contain annotations, but have the last channel as the stim channel (at least for BDF, that's the standard behavior of the BioSemi recording app). We could also set different default behavior for EDF, EDF+, and BDF separately (maybe that's a good compromise). GDF is also affected, but I think we don't need to change anything because I think the default is to parse events from the event table if present and do nothing otherwise. @larsoner @agramfort @teonbrooks?

raubsau, 2018-07-26:
@cbrnr Thank you for the reply. What about files with no stim channel but user-generated annotations (events marked as in `edf_test_edited.edf`)? These occur alot in my field... any chance to implement a read in with text annotations that are not stimulation information?

cbrnr, 2018-07-26:
Annotations are only possible with EDF+. They are stored in regular channels, which are parsed by MNE if `stim_channel='auto' (the default). Due to historic reasons MNE always creates a stim channel from these annotations, which means that text gets lost. However, you can always recover the original annotations using `raw.find_edf_events()`.

raubsau, 2018-07-26:
Thank you, I did not know that.

agramfort, 2018-07-26:
I don't have much time to comment but the annotations in edf files should
become raw.annotations attribute by default and not produce a stim channel
if not present in the file. Then it should be possible to get events from
raw.annotations.

See previous discussion on this.

It will speed up IO by avoiding extra allocations and heavily simply EDF IO
code, especially with preload=False

agramfort, 2018-07-27:
>
> I didn't follow the previous discussion regarding annotations closely, but
> (1) MNE has *always* produced a stim channel from annotations, and (2)
> annotations *are* stored in a channel (albeit a special channel) in EDF+.
>
> However, if you are thinking more in terms of bigger changes, then yes, we
> should treat events as annotations and not try to create a stim channel
> from them. So for files that do have a native analog stim channel, we still
> have to parse that and store the resulting events as annotations. For files
> that already have events (and no analog stim channel) (e.g. EDF+, GDF), we
> just store them as annotations without creating an artificial analog stim
> channel. Is there an issue/PR on this?
>
see https://github.com/mne-tools/mne-python/pull/4864

I should have opened a proper issue though

cc @massich

larsoner, 2019-08-21:
Closing as these files are openable now

closed 2019-08-21 by larsoner