mne-tools/mne-python #5346: Resizing after dropping flat epochs fails - does not own its data

stfnrpplngr, 2018-07-18:
Hi,

I get the following error:

```
4 bad epochs dropped
Traceback (most recent call last):

  File "<ipython-input-11-6faf2142abd0>", line 13, in <module>
    epochs.drop_bad(flat=dict(seeg=100e-10))

  File "<string>", line 2, in drop_bad

  File "c:\users\user\documents\github\mne-python\mne\utils.py", line 725, in verbose
    return function(*args, **kwargs)

  File "c:\users\user\documents\github\mne-python\mne\epochs.py", line 1053, in drop_bad
    self._get_data(out=False)

  File "<string>", line 2, in _get_data

  File "c:\users\user\documents\github\mne-python\mne\utils.py", line 725, in verbose
    return function(*args, **kwargs)

  File "c:\users\user\documents\github\mne-python\mne\epochs.py", line 1254, in _get_data
    data.resize((n_out,) + data.shape[1:], refcheck=False)

ValueError: cannot resize this array: it does not own its data
```

A similar case has been reported [here](https://github.com/mne-tools/mne-python/pull/2211). @larsoner provided a [fix](https://github.com/mne-tools/mne-python/pull/2211/commits/3b3a68b9591de1240da90f8e137473d0a05dfda7) back then.

stfnrpplngr, 2018-07-18:
A workaround:

`epochs = epochs.copy().drop_bad(flat=dict(seeg=100e-10))`

larsoner, 2018-07-18:
Can you make a minimal `gist` from `sample` or (better) `testing` data showing the problem?

stfnrpplngr, 2018-07-27:
I'm sorry for the delay:

```
import mne
from mne.datasets import sample

data_path = sample.data_path()
raw_fname = data_path + '/MEG/sample/sample_audvis_raw.fif'
raw = mne.io.read_raw_fif(raw_fname)
raw.load_data()

events = mne.find_events(raw)
epochs = mne.Epochs(raw, events, preload=True)

epochs.resample(512.)
epochs.crop(tmin=-0.25, tmax=1.25)
# This would be a workaround
# =============================================================================
# epochs = mne.EpochsArray(events=epochs.events, event_id=epochs.event_id,
#                          data=epochs.get_data(),
#                          info=epochs.info, tmin=epochs.tmin)
# =============================================================================

epochs.drop_bad(flat=dict(eeg=100e-8))
epochs.drop_bad(reject=dict(eeg=100e-5))
```

PR #5378 (2018-08-01): MRG: Fix resizing
Closes #5346.

@stfnrpplngr can you see if this fixes your problem?
closed 2018-08-01 by agramfort