mne-tools/mne-python #5347: plot error in plot_mne_inverse_label_connectivity.py

candleinwindsteve, 2018-07-18:
Hi, 

Can you please fix the bug in plot_mne_inverse_label_connectivity.py 

fig = plt.figure(num=None, figsize=(8, 4), facecolor='black')
no_names = [''] * len(label_names)
for ii, method in enumerate(con_methods):
    plot_connectivity_circle(con_res[method], no_names, n_lines=300,
                             node_angles=node_angles, node_colors=label_colors,
                             title=method, padding=0, fontsize_colorbar=6,
                             fig=fig, subplot=(1, 2, ii + 1))

plt.show()

It gives two separate plots rather than subplots in the same figure.

Thanks.
Feng

larsoner, 2018-07-18:
On my system I get two figures, the second of which has two subplots, which seems correct:

[image]

[image]

Is this not what you see?

larsoner, 2018-07-18:
@candleinwindsteve the images did not come through (they don't via email IIRC), can you go on GitHub and drag-drop them into a response?

larsoner, 2018-07-19:
Are you on the `master` version of MNE? We don't use `axisbg` anywhere anymore.

candleinwindsteve, 2018-07-19:
There are two warnings, do you think it is the first one that caused the problem? I feel it is not compatible with tight_layout that caused this issue. I am using the version that Martinos center has set up, I think probably I should install the latest version myself.

larsoner, 2018-07-19:
I don't think either warning caused the problem, but I do think they are indicative that you are using some older code somewhere that should be updated. In this case I suspect the MNE code you are using is old. What is your current `mne.__version__` and `mne.sys_info()`?

candleinwindsteve, 2018-07-19:
mne.__version__
Out[3]: '0.14.1'
mne.sys_info()
Platform:      Linux-3.10.0-693.21.1.el7.x86_64-x86_64-with-centos-7.4.1708-Core
Python:        2.7.13 |Anaconda custom (64-bit)| (default, Dec 20 2016, 23:09:15)  [GCC 4.4.7 20120313 (Red Hat 4.4.7-1)]
Executable:    /usr/pubsw/packages/python/anaconda2.7/bin/python2.7
mne:           0.14.1
numpy:         1.12.1 {lapack=mkl_intel_lp64, blas=mkl_intel_lp64}
scipy:         0.19.0
matplotlib:    2.0.2
sklearn:       0.18.1
nibabel:       2.1.0
mayavi:        Not found
pycuda:        Not found
skcuda:        Not found
pandas:        0.20.1

larsoner, 2018-07-19:
That is a couple of versions behind, you should probably update. You might want to use your own local install of Python, too.

candleinwindsteve, 2018-07-19:
Thanks, I wil set up my own env.

larsoner, 2018-07-19:
Okay, feel free to reopen if it still doesn't work with latest `mne`

closed 2018-07-19 by larsoner