mne-tools/mne-python #5349: Order of vertices in morphed source spaces

hichamjanati, 2018-07-18:
Hi, 

When morphing the source space of fsaverage to 'sample', the order of the morphed vertices is not preserved. The first 10 vertices of the morphed source space do not correspond to those of fsaverage:

fsaverage: 

[image]

Morphed fsaverage -> sample:

[image]

`import numpy as np  # noqa
import mne
import os

from mne.datasets.sample import data_path
from mayavi import mlab  # noqa
from surfer import Brain  # noqa

data_path = data_path()
subjects_dir = data_path + "/subjects/"
os.environ['SUBJECTS_DIR'] = subjects_dir

fname_fs = subjects_dir + "fsaverage/bem/fsaverage-ico-5-src.fif"
src_fs = mne.read_source_spaces(fname_fs)
src_sample = mne.morph_source_spaces(src_fs, subject_to="sample",
                                     subjects_dir=subjects_dir)
sources = np.arange(10)

for subject, src in zip(["fsaverage", "sample"], [src_fs, src_sample]):
    brain = Brain(subject, 'lh', 'inflated', subjects_dir=subjects_dir)
    surf = brain.geo['lh']
    vertices = src[0]["vertno"][sources]
    mlab.points3d(surf.x[vertices], surf.y[vertices],
                  surf.z[vertices], color=(1, 0, 0), scale_factor=6.)
    mlab.savefig("%s-sources.png" % subject)
`

This can be fixed by removing `np.sort` in https://github.com/mne-tools/mne-python/blob/master/mne/source_space.py#L2639 

       ` to['vertno'] = np.sort(best[fro['vertno']])`

When sorting the vertices one loses their correspondance which is the purpose of morphing. The resulting forward operators based on these morphed source spaces would have completely mismatched locations.

Written with @agramfort

jasmainak, 2018-07-18:
@hichamjanati can you ask you to open a PR? ;)

agramfort, 2018-07-18:
I think it's more complicated than this. I talked to @larsoner 

MNE-C also outputs sorted vertices and I don't know why. We need @mshamalainen wisdom on this. For me it defeats the purpose of source space morphing as with this sorting you still have to do stc.morph for group studies.

If we remove the np.sort then we might break stuff like backward compat with MNE-C and also some of our functions require stc.vertno to be sorted. Or at least if it's not you get a ValueError...

mmagnuski, 2018-12-19:
@hichamjanati You may find [this function](https://users.aalto.fi/~vanvlm1/conpy/generated/conpy.utils.get_morph_src_mapping.html#conpy.utils.get_morph_src_mapping) useful. Also @wmvanvliet might be interested in this issue.

closed 2018-12-19 by agramfort

mmagnuski, 2018-12-20:
@agramfort - Is this behavior fixed with the new morphing?

agramfort, 2018-12-20:
no but I am not sure we can fix it in a reasonable way.