mne-tools/mne-python #5356: Brainvision marker parser fails w/ [] characters

maedoc, 2018-07-20:
I have a (perhaps poorly chosen) marker in a file I received, 
```ini
Mk9=Comment,Discontinuit? [B] 4156 ms,136544,1,0
```
which at `brainvision.py:269` is being found with 
```python
items = re.findall(r"^Mk\d+=(.*)", mk_txt, re.MULTILINE)
```
but this regex stops completely at the `[` character, not finding the rest of the line nor the markers which follow.

I'm not a regex ninja but if I manage to fix, I'll send a PR.  In the meantime, I will change remove the offending characters in the file and move on.

sappelhoff, 2018-07-27:
Mmh are you sure this is due to the square brackets? Fiddling around a bit on [regex101](https://regex101.com/r/wubbJb/1) shows that the regex works as expected (at least on that line) ...

Could you share the `.vmrk`?

cbrnr, 2018-07-27:
Yes, `.` matches any character including square brackets.

jona-sassenhagen, 2018-07-30:
I think @teonbrooks wrote that code? :)

maedoc, 2018-07-31:
@sappelhoff indeed. I just realized that a few lines up from there, there's a regex for `r"\[.*\]"` that may be the culprit.

sappelhoff, 2018-08-06:
You are right @maedoc:
https://github.com/mne-tools/mne-python/blob/18755d22c3752b767dd493ffb4ffa5360bd322b4/mne/io/brainvision/brainvision.py#L305-L312

The `r"\[.*\]"` part is being used to find the line where a new section begins like `[Marker Infos]` or `[Common Infos]` 

Actually, this points to a very simple fix. Instead of `m = re.search(r"\[.*\]", mk_txt)` we can use `^` and `$` to get: `m = re.search(r"^\[.*\]$", mk_txt)`. This asserts that `[Marker Infos]` and other section headers are always written isolated and on a single line (which is true in BrainVision format). Then even your weird markers should be captured ;-)

see: #5412

PR #5412 (2018-08-07): bug fix #5356 scan for BrainVision sections using ^ and $
fixes #5356
commit 3b0270ca (2018-08-07): bug fix #5356 scan for BrainVision sections using ^ and $ (#5412)

* bug fix #5356 scan for BrainVision sections using ^ and $

* add comment why 'New Segment' markers are not parsed

* add test for square brackets in marker comments

* adjust test data to be more versatile
closed 2018-08-07 by jona-sassenhagen