mne-tools/mne-python #5357: Brainvision does not read meas_date properly

sappelhoff, 2018-07-23:
The `raw.info['meas_date']` is defined in the [docs](https://martinos.org/mne/dev/generated/mne.Info.html#mne.Info) to do this:

```
meas_date : list of int

    The first element of this list is a POSIX timestamp
 (milliseconds since 1970-01-01 00:00:00)
denoting the date and time at which the measurement was taken.
The second element is the number of microseconds.

```
First of all: Why would we need a list of
- POSIX miliseconds
- and then POSIX microseconds?

Second: When reading brainvision data, the following is executed concerning `raw.info['meas_date]`

https://github.com/mne-tools/mne-python/blob/241161d00d685c988a5937888f8c76959d0cd224/mne/io/brainvision/brainvision.py#L463

Was this perhaps intended as a temporary placeholder but it got forgotten? It is not a list and it does not contain the measurement date but the date at reading of the file.

The actual recording date can be obtained from the .vmrk file. See the documentation from Brain Products below:

[image]

So it usually gets written by the Brain Products equipment at the event called "New Segment". I can make a PR if you agree that the current code is strange.

larsoner, 2018-07-23:
`time.time()` is wrong, yes please read from the file if you can

larsoner, 2018-07-23:
> Why would we need a list of
>
> POSIX miliseconds
> and then POSIX microseconds?

Probably due to file storage specifications of FIFF

larsoner, 2018-07-23:
(microseconds are *in addition to* the specified number of milliseconds)

agramfort, 2018-07-24:
indeed this needs improvement.

please open a PR.

thanks

sappelhoff, 2018-07-24:
> (microseconds are in addition to the specified number of milliseconds)
This still does not make sense to me. See this example:

```python
# Parse BrainVision date
date_str = '20131113161403794232'
meas_date = datetime.strptime(date_str, '%Y%m%d%H%M%S%f')

# Convert to format for FIF meas_date
# We need list of unix time ... and unix time in microseconds
meas_date_unix = time.mktime(meas_date.timetuple())
meas_date_unix_micro = meas_date_unix * 1000000.
raw.info['meas_date'] = [meas_date_unix, meas_date_unix_micro]
```

This would give us `[1384355643.0, 1384355643000000.0]`. How does that help?

larsoner, 2018-07-24:
That looks like a bug

PR #5361 (2018-07-25): read meas_date from BrainVision vmrk
fixes #5357 

Note that I made some convenient changes to the test files that do not influence the other tests but allow us to test this new function without having to clutter more additional test files into `/tests/data`
closed 2018-07-25 by cbrnr