mne-tools/mne-python #5359: Some problem about importing .set file from EEGLAB

GalaChen, 2018-07-23:
Hello,
I want to use mne.io.read_raw_eeglab to import .set file from eeglab. However, it always report error with "No such file or directory: 'xxxx.set.mat". I can't understand why it append '.mat' to my .set file.
I have read the issues: [https://github.com/mne-tools/mne-python/issues/2672](url), it recommended some solution ways. Every method in that issue that I have tried didn't work in my computer. 
So, is there any other methods can solve my problem?

I used:
Python   2.7
MNE  0.16.2
scipy 0.19.1
MatLab  2013a

larsoner, 2018-07-23:
That is a bit odd, since we don't have this problem on our CIs. Can you try adding `appendmat=False` to the `loadmat` calls in `mne/io/eeglab/eeglab.py` to see if that fixes the problem? Eventually #5326 should take care of this problem.

GalaChen, 2018-07-23:
Thanks for help from @larsoner 

Follow your suggestion, I upgrade my eeglab.py according to [#5326](https://github.com/mne-tools/mne-python/pull/5326).
However, when I use mne.io.read_raw_eeglab(raw_fname), it reported like follow:
“TypeError: read_mat() got an unexpected keyword argument 'uint16_codec'.”

So, what I should do to fix it?

larsoner, 2018-07-24:
For now do not take `eeglab.py` from #5326. Just use your original `eeglab.py` and modify any `loadmat` calls to have `appendmat=False`

GalaChen, 2018-07-24:
I tied it and modify all `loadmat` in `mne/io/eeglab/eeglab.py`, but it doesn't work. 
Is there any other method to fix this problem?

larsoner, 2018-07-24:
Can you upload a small file somewhere and code snippet demonstrating the problem?

cbrnr, 2018-07-25:
This file contains epochs so you need to load it with `mne.io.read_epochs_eeglab`. Furthermore, the actual data contained in `1_LH.fdt` is missing.

GalaChen, 2018-07-25:
Yes, actual data contained `1_LH.fdt`
According to your advice, I used `mne.io.read_epochs_eeglab`, but it still report error about  
"No such file or directory: 'F:/MNE/MNE-Practice/Class-Copy/main-ERP1/data/1_LH.fdt.mat'"

So, how can I fix this error?

I'm a newer about EEGLAB and MNE, could you help and tell me how should I do with `1_LH.fdt`?

larsoner, 2018-07-27:
@GalaChen I have updated #5326, can you try using that bracnh?

GalaChen, 2018-08-02:
Platform:      Windows-post2008Server-6.2.9200
Python:        2.7 (r27:82525, Jul  4 2010, 07:43:08) [MSC v.1500 64 bit (AMD64)]
Executable:    D:\Program Files\Python 2.7\python.exe
CPU:           Intel64 Family 6 Model 58 Stepping 9, GenuineIntel: 4 cores
Memory:        Unavailable (requires "psutil" package)
mne:           0.16.2
numpy:         1.13.1 {lapack=mkl_lapack95_lp64, blas=mkl_lapack95_lp64}
scipy:         0.19.1
matplotlib:    2.2.2 {backend=TkAgg}
sklearn:       0.19.1
nibabel:       Not found
mayavi:        Not found
pycuda:        Not found
skcuda:        Not found
pandas:        0.22.0

This is what I get from the code. So, whether my scipy is too old?

agramfort, 2018-08-03:
@GalaChen no idea what's going on.

See if you can work with the dev version of MNE and fix the code to make it work on your system.

jona-sassenhagen, 2018-09-23:
Has the problem been resolved?

larsoner, 2018-10-04:
Closing due to non-response, can reopen if it's still a problem

dengemann, 2018-10-16:
Hi there! An attendee of the MNE workshop in Lyon just made me aware of this issue that is currently preventing her from switching to MNE-Python from EEGLAB :) Do we know if this has been fixed already? cc @larsoner @agramfort @GalaChen

larsoner, 2018-10-16:
Has she tried latest `master`? It's possible it has been fixed in `master`. I don't think anyone else has been able to reproduce

massich, 2018-10-16:
@dengemann if not fixed in master, create a toy example that reproduces what prevents her from switching and we'll make it happen.

thht, 2018-10-31:
just my 2cts:

if it boils down to the `TypeError: read_mat() got an unexpected keyword argument 'uint16_codec'` error, i am quite sure, it has been fixed in master with the new pymatreader version. if i recall correctly, @larsoner  proposed the fix because of this exact error.

larsoner, 2018-11-05:
Closing again since we still have not seen this reproduced

closed 2018-11-05 by larsoner

mmagnuski, 2018-11-18:
I had this error when trying to read for example `.raw` files with eeglab reader. :)

mmagnuski, 2018-11-18:
I also blamed old scipy at the beginning but then realised that what I am trying to read is not a .set file at all.