mne-tools/mne-python #5362: ENH: Ditching regularization - beamformer with rank-based inversion of cov matrix

britta-wstnr, 2018-07-24:
We have been experimenting with @sarangnemo with a beamformer variant that does not require the setting of a regularization parameter. Instead of regularizing the covariance matrix, it estimates the rank of the covariance matrix and sets the number of rank-deficient components to zero during the inversion of the covariance matrix. Thus, the enhancement of this method is that there is no "guessing" of the regularization parameter involved.

So far, we got very good results with this, especially also on **highly rank-deficient data like maxfiltered data.** 

A branch of my fork of MNE-Python with this beamformer variant implemented can be found here:
https://github.com/britta-wstnr/mne-python/tree/lcmv_cov_inv

And see here for the enhanced beamformer API (parameter ``eig_inv`` added to enable this variant):
https://github.com/britta-wstnr/mne-python/blob/lcmv_cov_inv/mne/beamformer/_lcmv.py#L26

I would be happy to make this a PR if there is an interest in that!

larsoner, 2018-07-24:
We do an operation like this for the whitening step in minimum norm inversion. It involves use of an optional `rank` parameter. It would be great if you could reuse code there as much as possible to do this. And it sounds like a great way to make the two operations have more similar options.

agramfort, 2018-07-24:
yes we should do this by default. as @larsoner says it's what is done in MNE/dSPM during the withening step.

cc @vlitvak who suggested the same thing.

we have our phantom data to test this.

britta-wstnr, 2018-07-26:
so far, I implemented it from scratch to be able to easily play around with it, if it goes into MNE-Python we should of course refactor! So thanks for the hints, I will look into it!

wmvanvliet, 2018-08-22:
It seems like a no-brainer to me to replace this ugly hardcoded regularization in the current beamformer implementation:

    Wk[:] = np.dot(linalg.pinv(Ck, 0.1), Wk)

with your approach @britta-wstnr.

agramfort, 2018-08-23:
if it's thoroughly tested no objection but we should avoid silently
changing people's results...

larsoner, 2018-11-05:
We plan to unify in 0.18 now

larsoner, 2021-02-19:
I think we have this now, `reg=0` gives sensical results in the tests at least I think.

closed 2021-02-19 by larsoner