mne-tools/mne-python #5364: How can I change the dtype of RawEDF?

PoiScript, 2018-07-25:
mne-python threw a memory error when I was trying to load a 3.5G edf file. Then I realized that it attempted to create a float64 array which was about 15G and caused the error.

So I wanna know how to change the `dtype` of `RawEDF` class? Data are stores as 2 bytes integer in EDF file. It just wastes memory to use float64.

cbrnr, 2018-07-25:
If you want to do any calculations with the data it is necessary to convert the 16bit integer values to floats (during this process, the data are also rescaled). You could try to use a memory-mapped file on disk by setting the `preload` argument to a file name.

PoiScript, 2018-07-26:
Sorry for the delayed response. Look like I was wrong in the frist place. I will try your suggestion instead. Thanks. Closing this one.

closed 2018-07-26 by PoiScript

PoiScript, 2018-07-26:
OK... I had tried to setup the edf file as a memory-mapped file, however I got an error as follow:

```python
f = open("xxx.edf", "r+b")
mm = mmap.mmap(f.fileno(), 0)
raw = mne.io.read_raw_edf("test.edf", preload=mm)
# ValueError: bad preload: <mmap.mmap object at 0x7f76c0d326a8>
```

And there is not document or example about using memory-mapped file, so I'm not sure which is the correct way to do so.

cbrnr, 2018-07-26:
I have never used memory-mapped files myself, but I think it should be something like `raw = mne.io.read_raw_edf("test.edf", preload="mmap.mm")`, which creates a file named `mmap.mm` (but you can choose a different file name if you want).

PoiScript, 2018-07-26:
@cbrnr Oh, thank for your quick response. Look like it's going to create a memory-mapped file for you and you don't have to do it manually.