mne-tools/mne-python #5365: ENH: agenda for beamformer module

britta-wstnr, 2018-07-26:
Let's keep record  of what issues and enhancements are still open with the beamformer code (*not* including visualization etc.). Here is a list of open issues and PRs and other things that need to be done:
 
 - [ ] **Projections in beamfomer code**
   Still open from #5135: how should DICS handle projections? Also related to #5342.
 - [ ] **[BUG] Orienation selection in unit-noise-gain DICS**
   Compare [this comment](https://github.com/mne-tools/mne-python/pull/5135/files#r200652209), the normalization done at the beginning of the DICS pipeline results in a wrong orientation selection formula
 - [ ] **Further refactoring of the DICS/LCMV code**
   Once the DICS bug (see above) is solved, the last steps of refactoring can be done (confirm equal results when switching DICS and LCMV computation blocks), this concerns [this code block (review).](   https://github.com/mne-tools/mne-python/pull/5135/files#r182999817)
   At the moment, results do not converge, probably due to the DICS orientation selection bug. 
- [ ] **Orienation selection for unit-gain LCMV**
  This is started in #4659 and is waiting for easy access to leadfield normalization, which should be granted after merging the refactoring PR #5135.
- [ ] **Pass covariance to beamformer to get static image**
  This is issue #5229. It relates ``_lcmv_source_power``, which could need some work / refactoring with ``make_lcmv``.
- [ ] **Eigenvalue inversion beamformer ("automatic" regularization)**
  This is issue #5362.
 
And then, there is this older issue about beamformer enhancements: #3853. Some of this is done now, but other things might remain, like the different output types (*z-scores* etc. - which, however, could also just be computed on the output directly).

Anything missing, @agramfort @larsoner @sarangnemo @wmvanvliet ?

larsoner, 2018-07-26:
Sounds reasonable to me

wmvanvliet, 2018-07-27:
Yeah, I agree on all these points.

Regarding projections with DICS: turns out, applying SSPs to a CSD works exactly the same as applying them to a covariance matrix! See this gist: https://gist.github.com/wmvanvliet/fe04c40e27c8f81422596163d41563ca

britta-wstnr, 2018-08-01:
Added storing of source orientations in ``filters``.
cc @sarangnemo

britta-wstnr, 2018-08-07:
Added speeding up LCMV
as brought up by @larsoner in #5135

britta-wstnr, 2018-08-13:
@larsoner I added the combination of different sensor types.

larsoner, 2018-09-20:
@britta-wstnr the max-power has a sign ambiguity. Any reason not to orient it so that positive is in the direction of the vertex normal? Basically this means taking the max power orientation and multiplying by the sign of the dot product with the vertex normal. It's a principled way to deal with the arbitrary flip.

britta-wstnr, 2018-09-25:
@larsoner How would we do that in the volume case, do volume fwd models have that information as well? (Note that beamformers are usually computed for the full source grid, not a surface solution.)

larsoner, 2018-09-25:
The normals are defined as up Z in that case IIRC. So it will still work

wmvanvliet, 2018-11-08:
> @wmvanvliet can you take a look at the list above and update the checkboxes?

done!

mmagnuski, 2018-12-17:
(I just moved unchecked checkboxes to the top to make them more visible)

larsoner, 2019-09-11:
Clearing 0.19 milestones related to beamformer, @britta-wstnr let me know if there were things you wanted to get into 0.19