mne-tools/mne-python #5366: plot always behaves as if clim 'kind' is 'percent' even when 'value' is specified

nrafidi, 2018-07-26:
The following code snippet:

`mne.viz.plot_source_estimates(src, subject=STRUCTURAL, background=bk, surface='inflated', hemi='lh', views='lat', clim={'kind': 'value', 'lims': lims}, colormap=cmap, subjects_dir=SUBJ_DIR, smoothing_steps=smoothing_steps, spacing='ico4', backend='matplotlib')`

Is faulty. Even though clim 'kind' is specified as 'value', the resulting plot is scaled as if 'kind' were 'percent'

I believe the bug is on this line:

https://github.com/mne-tools/mne-python/blob/7e0e988b6e8f3461855a1dd36aec7e5d17b6c379/mne/viz/_3d.py#L1368

Independent of clim specifications, this line scales the colormap by the maximum value of the to-be-plotted values, which is what gives the error. In my local copy, when I change it to colors = array_plot, the plots become correct.

agramfort, 2018-07-26:
thanks @nrafidi 

can you open a PR?

larsoner, 2018-10-04:
This has been fixed

closed 2018-10-04 by larsoner