mne-tools/mne-python #5368: Missing tmax parameter in the constructor of EpochsArray?

underscoredam, 2018-07-26:
In the constructor of `Epochs` class, there are both `tmax` and `tmin` parameters. However, in the constructor of `EpochsArray` class, there is no `tmax` parameter (although the `reject_tmax` parameter exists in both constructors). Is this a bug? If it is, I can submit a pull request.

This question arose when I wanted to play with a subset, a window, of epochs. I have an `EpochsArray` instance, `master`, with `master._data.shape=(60,33,4000)`. I wanted to obtain a new `EpochsArray` instance , `window`, having `window._data.shape=(60,33,200)` or something like that. I was thinking of using something like following:
```
window = EpochsArray(master._data, info, tmin=0, tmax=1.0, events=parent.events)  # sfreq=200.0
```
Is this how I am supposed do it, or is there a different approach to this? Please note that I will need to keep the `parent` untouched. Thanks!

larsoner, 2018-07-26:
given `tmax`, `info['sfreq']`, and `data.shape[-1]` the `tmax` is implied. If you want a different window, contsruct the `EpochsArray` then do `epochs.crop`

closed 2018-07-26 by larsoner

mmagnuski, 2018-07-29:
@underscoredam: just to clarify - Eric meant: "given **tmin**, ..."