mne-tools/mne-python #5369: mne.SourceEstimate.plot() Graph Unresponsive

corscheid, 2018-07-26:
I followed the installation and setup tutorial given at:
https://www.martinos.org/mne/stable/getting_started.html

Then I downloaded and tried to run this full example script given at:
https://martinos.org/mne/stable/_downloads/plot_mne_dspm_source_localization.py

It runs to the end, but the five brain plots that pop up are instantly unresponsive. 

Here is my sys_info() output:
```
Platform:      Darwin-17.7.0-x86_64-i386-64bit
Python:        3.6.6 |Anaconda, Inc.| (default, Jun 28 2018, 11:07:29)  [GCC 4.2.1 Compatible Clang 4.0.1 (tags/RELEASE_401/final)]
Executable:    /anaconda3/envs/mne/bin/python
CPU:           i386: 8 cores
Memory:        16.0 GB

mne:           0.16.2
numpy:         1.14.5 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.1.0
matplotlib:    2.2.2 {backend=module://ipykernel.pylab.backend_inline}

sklearn:       0.19.1
nibabel:       2.3.0
mayavi:        4.6.1 {qt_api=pyqt5}
pycuda:        Not found
skcuda:        Not found
pandas:        0.23.1
```

There are no error messages given, which makes this much more difficult to debug. Any help would be much appreciated!

larsoner, 2018-07-27:
Can you try using `%matplotlib qt` and/or `%gui qt`? It starts up an event loop that should also process the Mayavi events

corscheid, 2018-07-27:
Hi @larsoner 

Running `%matplotlib qt` from the Ipython console prevents the freeze.  I also noticed that changing Spyder settings for IPython console Graphics backend from `Inline` to `Qt5` has the same effect as running `%matplotlib qt` but it's persistent.

The buttons at the top of the plot window don't respond or do anything, but even so, at least the plot isn't unresponsive and I can drag the 3D image around and have a good look at it. Much appreciated!

larsoner, 2018-08-09:
Reopening as we should update our install docs to make a (better) note of these sorts of tweaks

corscheid, 2018-09-06:
Thanks @larsoner for your efforts and replies. My colleague @deepagupta would be happy to reply here in the event of further questions or discussion regarding documentation or related MNE functions or structures we are using.

larsoner, 2018-10-06:
It looks like we already have the suggestion of using `%gui qt` in the dev docs so this can be closed

closed 2018-10-06 by larsoner