mne-tools/mne-python #5370: mayavi issues with vtk and tvtk

massich, 2018-07-27:
I was following our installation instructions and I got into trouble with vtk and tvtk. See the following snipped:

```sh
sed -e 's/name: mne/name: mne_mayavi_test/' environment.yml > /tmp/mne_mayavi_test.yml
conda env create -f /tmp/mne_mayavi_test.yml 
source activate mne_mayavi_test
echo "from mayavi import mlab\nmlab.test_plot3d()\nmlab.savefig('/tmp/xx.png')" > /tmp/mayavi_test.py
ipython --gui qt /tmp/mayavi_test.py
```

we get this warning: 
```sh
********************************************************************************
WARNING: Imported VTK version (8.1) does not match the one used
         to build the TVTK classes (7.1). This may cause problems.
         Please rebuild TVTK.
********************************************************************************
```

and finally crashes: 
```sh
, trait: aa_frames, old value: 8, new value: 0
Traceback (most recent call last):
  File "/home/sik/miniconda3/envs/mne_mayavi_test/lib/python3.6/site-packages/traits/trait_notifiers.py", line 340, in __call__
    self.handler( *args )
  File "tvtk_classes/render_window.py", line 206, in _aa_frames_changed
    self._do_change(self._vtk_obj.SetAAFrames,
AttributeError: 'vtkRenderingOpenGL2Python.vtkXOpenGLRenderWindow' object has no attribute 'SetAAFrames'
```

mmagnuski, 2018-08-21:
Just BTW - I get this AttributeError with mne env on Windows when saving figures (but the figure is saved and looks ok).

jhouck, 2018-09-04:
By any chance was there a solution for this?  I get the same error in a freshly-installed mne env, both on my creaky centos 6 box and on an ubuntu 18.04 LTS VM.  The mlab figure itself looks fine, but `mlab.screenshot` give me an array of zeroes and `mlab.savefig` returns the AttributeError.  For me, at least, reinstalling mayavi has done nothing.   The error turns up consistently using the plot_decimate_head_surface example, although in spyder mlab.savefig writes a blank figure.

larsoner, 2018-09-05:
Can you try installing / upgrading to latest `mayavi` master? Maybe there is some bug fixed there. But the fact that the figure itself looks fine but your screenshot looks bad does not bode well for Mayavi fixing the problem. One thing you could do is try using `offscreen` rendering mode if you are not trying to do interactive viz.

jhouck, 2018-09-07:
Yes, this was in mayavi 4.7.0.dev0, although 4.6.1 did the same thing.  After reinstalling, things work in the VM using offscreen rendering.  The centos box has some ongoing issues.  I'd used miniconda for both to save disk space, so might try again on centos with anaconda, on the outside chance that there might be some hardware-specific missing dependency.

jhouck, 2018-09-07:
Ananconda (vs. miniconda) seems to have done the trick.  Things now work with offscreen rendering on the creaky old centos box.

larsoner, 2018-11-07:
Closing as this seems like an installation / `ENABLE_USER_SITE` problem that presumably has been fixed (?). Feel free to open if it's still a problem or we need to update our docs somehow

closed 2018-11-07 by larsoner