mne-tools/mne-python #5375: MNE Coreg fiducials disappear

genisprat, 2018-07-30:
We are using MNE coreg to mark the fiducial locations in MRI images. We observe some strange behavior whenver one of the fiducials is in front of the head. In this case the fiducial disappears and is not visible anymore. To illustrate this with this screenshots: 

[image]
Here all three fiducials are set, but only the lpa and rpa are have "open space" behind them and are not occluded.
[image]
Rotating the view such that the head is behind all fiducials ocludes all of them.

[image]

Turning the view even more makes the nasion visible (at least those parts that are not in front of the head).

We run the latest MNE version (from master, python 3.6) on a ubuntu (16) linux machine where we have followed the MNE install instructions. Our VTK version is 8.1.1. The error also occurs after updating mayavi to the latest version from master. 

Any ideas how to fix this? E.g. different VTK versions?

larsoner, 2018-07-30:
We draw the *inside* of the head opaque at all times. I suspect the *outside* of the head surface for you is fully transparent (the transparency is controllable). Can you cehck in `Display options...` if you have the head opacity at or near 0?

This behavior also manifests as the head turning "the wrong direction" when you click-drag to rotate.

genisprat, 2018-07-30:
Thanks for the quick reply! This behavior does not change when we change the opacity slider in the display options. So fiducials disappear across the full range of opacity values.

larsoner, 2018-07-30:
If you have time, you could click the somewhat-triangular icon in the upper left corner of the display area. This is the Mayavi pipeline editor. You should be able to tweak params of the surfaces we display (there should be two, one for inside and one for outside the head).

I'm not sure offhand what would cause this. I would try changing the visibility of the inside- and outside-of-head surfaces and see which combinations work and don't work. In theory setting both to have opacity 0.99 or so (instead of 1.) might also "fix" it.

genisprat, 2018-07-30:
Yes! Setting both to .5 makes them visible, but they are still occluded by both surfaces. Setting only one to 0.5 does not work. But this is definetly workable! Thank you very much!

larsoner, 2018-07-30:
Does setting both to 0.99 work? Sometimes with an actual value of `1.` Mayavi messes things up.

genisprat, 2018-07-30:
No, that doesn't work... We are happy to help by trying out more suggestions if anything comes to mind!

larsoner, 2018-07-30:
So in the front-left view (second screenshot you posted), you don't see any of the nasion or LPA?

genisprat, 2018-07-30:
Yes, even though all screenshots are taken from the same scene, so nasion and LPA should be visible.

larsoner, 2018-08-18:
So far I have been unable to replicate this. @genisprat can you give the output of `>>> mne.sys_info()` and also `$ glxinfo | grep version`?

genisprat, 2018-08-21:
>>> mne.sys_info()
Platform:      Linux-4.15.0-30-generic-x86_64-with-debian-stretch-sid
Python:        3.6.6 |Anaconda, Inc.| (default, Jun 28 2018, 17:14:51)  [GCC 7.2.0]
Executable:    /home/gprat/anaconda2/envs/mne/bin/python
CPU:           x86_64: 4 cores
Memory:        3.8 GB

mne:           0.17.dev0
numpy:         1.14.5 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.1.0
matplotlib:    2.2.2 {backend=agg}

sklearn:       0.19.1
nibabel:       2.3.0
mayavi:        4.6.2.dev0 {qt_api=pyqt5}
pycuda:        Not found
skcuda:        Not found
pandas:        0.23.3

glxinfo | grep version
server glx version string: 1.4
client glx version string: 1.4
GLX version: 1.4
    Max core profile version: 4.2
    Max compat profile version: 3.0
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.0
OpenGL core profile version string: 4.2 (Core Profile) Mesa 17.2.8
OpenGL core profile shading language version string: 4.20
OpenGL version string: 3.0 Mesa 17.2.8
OpenGL shading language version string: 1.30
OpenGL ES profile version string: OpenGL ES 3.0 Mesa 17.2.8
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.00

larsoner, 2018-08-21:
@genisprat are you using saftware rendering, then? If so, can you try using `llvmpipe` rather than `swr` (assuming that's what you're running)? You can check with `glxinfo | grep renderer`

Diptyajitd7, 2018-09-13:
@larsoner I have recently faced some similar problems.
##################
mne.sys_info() output:
Platform:      Linux-4.15.0-33-generic-x86_64-with-debian-buster-sid
CPU:           x86_64: 4 cores
Memory:        15.6 GB
mne:           0.17.dev0
numpy:         1.15.1 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.1.0
matplotlib:    2.2.2 {backend=Qt5Agg}
sklearn:       0.19.1
nibabel:       2.3.0
mayavi:        4.6.2 {qt_api=pyqt5}
cupy:          Not found
pandas:        0.23.4
dipy:          0.14.0
############### 
glxinfo | grep version output:
server glx version string: 1.4
client glx version string: 1.4
GLX version: 1.4
Max core profile version: 4.5
Max compat profile version: 3.0
Max GLES1 profile version: 1.1
Max GLES[23] profile version: 3.2
OpenGL core profile version string: 4.5 (Core Profile) Mesa 18.0.5
OpenGL core profile shading language version string: 4.50
OpenGL version string: 3.0 Mesa 18.0.5
OpenGL shading language version string: 1.30
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 18.0.5
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20

#############################################
Once I try to change the position of the fiducials, the coordinates are not updated except for the LPA. However, if I switch from RPA to LPA or Nasion and to switch back to RPA again, then coordinates for RPA are updated. Its also the same for Nasion. Fiducials are still missing but I can see them if I turn the head model (see the pic below).   [image]

I tried to tweak mayavi pipeline according to your comment, but I can't find any visibility option.  Am I looking into the wrong settings?  (pics below).
[image]
## display options: 
[image]

Last but not least, whenever I try to load a different subject file into mne coreg, the display automatically closes. This is the error message that I am getting in the terminal: (Error!)  'malloc(): smallbin double linked list corrupted'. Not sure why I am getting this error.  

Any help would be appreciated. Thanks!

ezemikulan, 2018-10-11:
Hi all, I had a similar problem and I fixed it by [installing the NVIDIA drivers](https://linuxconfig.org/how-to-install-the-nvidia-drivers-on-ubuntu-18-04-bionic-beaver-linux). Hope it helps.

bloyl, 2018-10-24:
I am having this issue as well

> >>> mne.sys_info()
> /opt/anaconda3/envs/mne-devel-py3/lib/python3.6/site-packages/sklearn/externals/joblib/externals/cloudpickle/cloudpickle.py:47: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
>   import imp
> Platform:      Linux-4.13.0-46-generic-x86_64-with-debian-stretch-sid
> Python:        3.6.6 |Anaconda, Inc.| (default, Oct  9 2018, 12:34:16)  [GCC 7.3.0]
> Executable:    /opt/anaconda3/envs/mne-devel-py3/bin/python
> CPU:           x86_64: 4 cores
> Memory:        19.0 GB
> 
> mne:           0.17.dev0
> numpy:         1.15.3 {blas=mkl_rt, lapack=mkl_rt}
> scipy:         1.1.0
> matplotlib:    3.0.0 {backend=Qt5Agg}
> 
> sklearn:       0.20.0
> nibabel:       2.3.1
> mayavi:        4.6.2 {qt_api=pyqt5, PyQt5=5.9.2}
> cupy:          Not found
> pandas:        0.23.4
> dipy:          0.14.0

> 
> (mne-devel-py3) bloyl@calisto:~$ glxinfo |grep version
> server glx version string: 1.4
> client glx version string: 1.4
> GLX version: 1.4
>     Max core profile version: 4.5
>     Max compat profile version: 3.0
>     Max GLES1 profile version: 1.1
>     Max GLES[23] profile version: 3.2
> OpenGL core profile version string: 4.5 (Core Profile) Mesa 17.2.8
> OpenGL core profile shading language version string: 4.50
> OpenGL version string: 3.0 Mesa 17.2.8
> OpenGL shading language version string: 1.30
> OpenGL ES profile version string: OpenGL ES 3.2 Mesa 17.2.8
> OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
>

bloyl, 2018-10-25:
Please note that this also effects digitization points.

PR #5666 (2018-10-29): MRG: Improve OpenGL compat
This PR:

- Adresses #5375 for graphics renderers (Intel, MESA, ...) that do not properly do Z-buffer based rendering (argh) by making it so the fiducials (and other items) are visible, even though they still aren't rendered perfectly.
- Improves Mayavi rendering by using full-screen anti-aliasing by default.

@bloyl can you see if it makes `mne coreg` usable for you with Intel/MESA rendering?
```
$ 
mne coreg -d ~/mne_data/MNE-sample-data/subjects -s sample --fif ~/mne_data/MNE-sample-data/MEG/sample/sample_audvis_raw.fif
```
After runinng this and clicking "Fit Fid [... cut]
closed 2018-10-29 by larsoner