mne-tools/mne-python #5382: Memory-Mapping Preload Fails When Adding Channels

alexrockhill, 2018-07-31:
When data is memory-mapped to a file, there is a memory error in:

e-packages/mne/channels/channels.py", line 877, in add_channels
    data = np.concatenate(data, axis=con_axis)
MemoryError

jasmainak, 2018-07-31:
A [minimal reproducible example](http://sscce.org/) on [gist](https://gist.github.com/) would be very much appreciated

alexrockhill, 2018-08-01:
https://gist.github.com/alexrockhill/f2ddbae467e8e5728db48834f2601e0d

Here is a gist that could work. There is some difficulty because the memory error depends on the RAM of the machine and also to create a RawArray, one must be able to load the RawArray into memory, so there must be a workaround where many raw files are made and loaded into memory-mapped files.

larsoner, 2018-08-07:
We could probably handle this more gracefully. It might be possible to resize the memmapped array. If not, then we could make a new one, possibly via param.

PR #5595 (2018-10-09): MRG: Resize memmap when adding channels
Closes #5382.

@alexrockhill can you see if it fixes your issue?
closed 2018-10-09 by agramfort