mne-tools/mne-python #5383: mne.SourceEstimate.data array

corscheid, 2018-07-31:
Hi, I really really hate to post general questions on GitHub Issues, but I don't know where else to go. Is there an MNE Python forum? I tried to join the mailing list but I don't think it was successful.

My current question is, what exactly is stc.data?

I know that it's a 2D array of vertices x time and that vertices could be voxels. But I'm not sure if the float values in that array are current density, and what are the XYZ coordinates for each of the 8000+ vertices if they really are voxels. If they are not voxels, then what are they? What units are the values found in stc.data?

We're working on processing EEG data using sLORETA method. We would like to move to MNE Python since using LORETA-KEY and EEGLAB is a pretty slow and manual process. What we’re interested in is getting the xyz coordinates and current density data as well as lobe location, just like the sLORETA Viewer shows, in an automated way. Is there any way we can do this in MNE?

Thanks,
Corey

larsoner, 2018-08-01:
As I mentioned on the mailing list, maybe a good start would be `plot_object_sourceestimate.py` that talks about:

1. what `.data` includes
2. what `.vertices` are
3. how these index a high-res surface
4. how to get the corresponding coordinates in Freesurfer RAS, MRI RAS, etc.
5. example of upsampling from these to the high-res surface

@corscheid would that help? Are there other things that would be good to add?

Then once this is in, eventually we could expand it to include `VolSourceEstimate` and finally `MixedSourceEstimate`.

corscheid, 2018-08-01:
@larsoner Yes, I think that would be very helpful. I'm not 100% sure what else would be good to add off the top of my head right now, but the more EEG-specific documentation (particularly from raw EEG data to source localization data and plotting) there is, the better, I think. I also spent a good bit of time wondering what a .fif file is, where they are expected to come from, and how all that works, coming from an EEGLAB/ERPLAB/sLORETA Viewer workflow.

Outside of using the sample data set and following the available MEG tutorials on the website, I'm a little bit lost in trial and error piecing everything together. Our workflow currently includes .cnt files and .set/.fdt and .txt files from EEGLAB and ERPLAB. My curiosity has been how to implement and automate all of these same steps in MNE Python for EEG data of a somewhat large set of subjects.

Thank you so much for your quick replies and patience!

agramfort, 2018-08-01:
maybe add these info here http://martinos.org/mne/stable/faq.html ?

larsoner, 2018-08-01:
I think there is too much information to cover for it the be appropriate for the FAQ

agramfort, 2018-08-01:
then add a page in section on "data structures and containers" in the
documentation.

larsoner, 2018-08-01:
My preference would be a `plot_object_sourceestimate.py` like we have for raw/epochs/evoked, or a `plot_background_freesurfer.py`. I fear "data structures and containers" is too general / broad.

agramfort, 2018-08-02:
@TommyClausner I think you have now all the knowledge to explain this.

can you take a stab at this? while you wait for us to review the other PR...

TommyClausner, 2018-08-02:
Hey @agramfort I appreciate your optimism ("all the knowledge to explain this") ;)

What I would do now is:
- making ``plot_object_sourceestimate.py`` (guess this might work)
- making ``plot_background_freesurfer.py`` - I know and used FreeSurfer a bit outside mne, however I have to dive a bit more into how exactly the steps in mne are - or are you really just interested in FreeSurfer and only FreeSurfer (like using "recon-all" and stuff) - and where which relevant data come from? (e.g. surf folder, labels folder, etc)

larsoner, 2018-08-02:
For FS just the parts we use, like high res mesh, decimated mesh, downsampling, upsampling, labels, annotations.

agramfort, 2018-08-03:
for plot_object_sourceestimate i imagine you explaining the logic of
stc.vertices and stc.data and how it relates to src[0]['rr'] to have
locations.

corscheid, 2018-09-06:
Thanks everyone for your efforts and replies. My colleague @deepagupta would be happy to reply here in the event of further questions or discussion regarding documentation or related MNE functions or structures we are using.

larsoner, 2018-10-04:
We have a bit of this in `master` now so I'm moving this to 0.18. @jasmainak if you feel like improving the `plot_object_stc` tutorial this issue has other concepts in it that should be included

larsoner, 2019-05-13:
@drammock is this one taken care of? If not, let's add to the documentation TODO list

drammock, 2019-05-15:
I would say that this is not yet fully resolved.  But I'll cross-link it to #5944, add a note to the roadmap there, and close this one.

closed 2019-05-15 by drammock