mne-tools/mne-python #5384: ref_meg trouble

jona-sassenhagen, 2018-07-31:
My MEG data says this now with essentially anything I do:

```
~/tools/mne-python/mne/io/pick.py in pick_info(info, sel, copy)
    406                     'compensation channels are missing: %s\n'
    407                     'Either remove compensation or pick compensation '
--> 408                     'channels' % (current_comp, comps_missing))
    409             else:
    410                 logger.info('Removing %d compensators from info because '

RuntimeError: Compensation grade 3 has been applied, but compensation channels are missing: ['BG1-3609', 'BG2-3609', 'BG3-3609', 'G11-3609', 'G12-3609', 'G13-3609', 'G22-3609', 'G23-3609', 'P11-3609', 'P12-3609', 'Q11-3609', 'Q12-3609', 'Q13-3609', 'R12-3609', 'R13-3609', 'R22-3609', 'R23-3609']
Either remove compensation or pick compensation channels
```

... what? Happens, for example, when I do `raw.pick_types(meg=True, ref_meg=False).
I can `raw.apply_gradient_compensation(0)`, but then my data looks kind of shitty.

jona-sassenhagen, 2018-07-31:
This works:

```
picks = mne.pick_types(raw.info, meg=True, ref_meg=False)
mne.viz.plot_topomap(raw._data[picks], mne.pick_info(raw.copy().apply_gradient_compensation(0).info, picks));
```

But it feels very wrong. And it doesn't work for many functions.

larsoner, 2018-07-31:
We have had a number of these. Should be fixed by #5342, feel free to comment on that proposal

agramfort, 2018-08-01:
@jona-sassenhagen what is a gist that allows to replicate this issue? you have a number of tiny CTF files in the mne-testing-data folder.

agramfort, 2018-08-04:
@jona-sassenhagen a fix along these lines https://github.com/mne-tools/mne-python/pull/5388

please share a gist to know how we can fix this.

larsoner, 2018-10-04:
@jona-sassenhagen can you write a minimal example from a `testing` or `mne/io/data/*.fif` file? It will make it easy for us to test that fixes/workarounds from #5342 work. It would be good to get to this example in the next week so a fix can be made the next week to make it into 0.17.

larsoner, 2018-10-08:
@jona-sassenhagen any chance to make this `gist` quickly? There is a PR open to take care of most of these problems and it would be great to have this test case

larsoner, 2018-10-08:
Okay actually I think this does it:
```
import os.path as op
import mne
raw = mne.io.read_raw_fif(op.join(op.dirname(mne.__file__), 'io', 'tests',
                                  'data', 'test_ctf_comp_raw.fif'))
evoked = mne.EvokedArray(raw[:, :10][0], raw.info)
evoked.pick_types(meg=True, ref_meg=False)
evoked.plot_topomap()
```
It fails on `master` but passes on the open PR, fortunately!

PR #5581 (2018-10-12): [MRG] Warn when removing channels needed to compensation matrices.
This is converts the RuntimeError that was raised when removing channels that were needed for applied compensation matrices to a Warning.

this is an initial PR in response to issue: #5342

If there are any other PRs that were made to skip this issue please add it to this list.

PRs to check
- [ ] #5125
- [x] #5133
- [ ] #5225
- [ ] #5270 
- [ ] #5295
- [x] #5388 
- [x] #5321
- [x] #5341

Closes #5384.
commit 46f3fa12 (2018-10-12): [ENH] Warn when removing channels needed to compensation matrices. (#5581)

This is converts the RuntimeError that was raised when removing channels that were needed for applied compensation matrices to a Warning.

this is an initial PR in response to issue: #5342

Closes #5384.
closed 2018-10-12 by massich