mne-tools/mne-python #5400: Allow `@` queries for metadata

jona-sassenhagen, 2018-08-04:
Metadata-based querying of Epochs is very powerful. For example, these work:

```
import mne
path_kword = mne.datasets.kiloword.data_path() + '/kword_metadata-epo.fif'
kword = mne.read_epochs(path_kword)
kword["NumberOfLetters == 4 & ConsonantVowelProportion < .5"], kword["WORD == 'area'"]
```

But what doesn't work is pulling a variable from the current scope with `@`:

```
word = 'area'
kword.metadata.query("WORD == @word")
```

This works, in Pandas. But this errors:

```
kword["WORD == @word"]
```

Would be cool to have!

@choldgraf

jasmainak, 2018-08-08:
wouldn't this be redundant with PyBIDS ?

jasmainak, 2018-08-08:
sorry my bad. Read too quickly

jona-sassenhagen, 2018-08-09:
>wouldn't this be redundant with PyBIDS ?

I'm kinda wondering how you misread this now ...

larsoner, 2018-08-09:
Probably living with too many open GitHub tabs or emails open, and clicking into one assuming it was the one you meant to respond to without actually checking. At least that's how it has happened to me in the past, too :)

jona-sassenhagen, 2018-08-11:
I tried to understand how Pandas does it, and I don't. And a self-made workaround based on looking up stuff in locals probably wouldn't work either.

Why are we using `pd.DataFrame.eval` btw over `.query`?

larsoner, 2018-08-11:
It was a while ago but I think we did have some discussion about which one to use. Maybe one is/was deprecated?

choldgraf, 2018-08-12:
I'm pretty sure that they accomplish mostly the same thing, but that `eval` returns a boolean mask according to whatever the query was, while `query` returns the DataFrame object itself. So probably we use `eval` because we want that mask somewhere else in the code, I'm guessing.

drammock, 2020-11-24:
... 2 years later, I tried to get this to work by manipulating the `level` kwarg (how far up the call stack pandas looks for `@variable`s), as well as passing `local_dict=globals()` and a few other hackish things; nothing worked.  Given that it's possible to generally replace use of `@` with a fairly similar f-string nowadays (e.g., `epochs['WORD in @words']` becomes `epochs[f'WORD in {words}']`) I think we can close this as YAGNI.

closed 2020-11-24 by drammock