mne-tools/mne-python #5401: ica.plot_properties only plots 3 epochs with raw input?

jona-sassenhagen, 2018-08-04:
Try:

```
import mne
path_eeglab = mne.datasets.testing.data_path()

fname = path_eeglab + '/EEGLAB/test_raw.set'
montage = path_eeglab + "/EEGLAB/test_chans.locs"

event_id = {"rt": 1, "square": 2}
raw = mne.io.read_raw_eeglab(fname, event_id=event_id, preload=True, montage=montage)
ica = mne.preprocessing.ICA(20, random_state=0).fit(raw.filter(1, None))
ica.plot_properties(raw, 0)
```

Gives:

[image]

I.e., for some reason, only 3 "epochs" are plotted.

@mmagnuski

agramfort, 2018-08-05:
I can replicate the pb.

mmagnuski, 2018-08-05:
I can replicate the problem too.
But it gets even weirder for me: each time I restart the notebook I get different axes filled with spectrum, epoch variance, etc. Below three examples:  

[image]
[image]
[image]

I currently have no other explanation than that your code is haunted...

agramfort, 2018-08-05:
I would try with an older matplotlib so see if we broke it or they did.

jona-sassenhagen, 2018-08-05:
Hm ... The axes are created here, and later unpacked order:

```
def _create_properties_layout(figsize=None):
    """Create main figure and axes layout used by plot_ica_properties."""
    import matplotlib.pyplot as plt
    if figsize is None:
        figsize = [7., 6.]
    fig = plt.figure(figsize=figsize, facecolor=[0.95] * 3)

    axes_params = {'topomap': [0.08, 0.5, 0.3, 0.45],
                   'image': [0.5, 0.6, 0.45, 0.35],
                   'erp': [0.5, 0.5, 0.45, 0.1],
                   'spectrum': [0.08, 0.1, 0.32, 0.3],
                   'variance': [0.5, 0.1, 0.45, 0.25]}
    axes = [fig.add_axes(pos, label=name) for name, pos in axes_params.items()]

    return fig, axes
```

Dict order should be fixed in Py3.6+, but is not guaranteed otherwise right?

Which would however not explain why it does not work on my Py3.6 either.

jona-sassenhagen, 2018-08-05:
(Also I think I wrote that code :()

jona-sassenhagen, 2018-08-05:
Oh! I think I got it then.

PR #5404 (2018-08-06): MRG fix order of plot_ica_properties for <py36
Closes #5401

@mmagnuski
closed 2018-08-06 by mmagnuski