mne-tools/mne-python #5408: Allow reordering channels?

jona-sassenhagen, 2018-08-05:
I have a multicenter dataset where the channels are not in the same order across datasets. It appears to me to be rather complicated to get them all into the same order, keep channel locations aligned, etc. Should we make this easier?

agramfort, 2018-08-05:
why is this needed? MNE should handle this transparently.

jona-sassenhagen, 2018-08-05:
For example, I can't do a grand average.

agramfort, 2018-08-05:
then grand_average should be fixed.

mmagnuski, 2018-08-05:
`raw.reorder_channels(channel_order)` doesn't work for you?

jona-sassenhagen, 2018-08-05:
What! I didn't know that existed. Still, mne.combine_evoked should do it behind the scenes.

mmagnuski, 2018-08-05:
Not that long ago `pick_channels` would reorder, but @larsoner changed its behaviour to not reorder, only pick. In that PR I asked him for similar reasons you present here to still allow to reorder and he added `reorder_channels`. It is more explicit this way, I like it.
If grand does not perform some safety checks for channel order it should. It does check other types of consistency like sfreq or times IIRC so why not chan order.

agramfort, 2018-08-06:
I think that grand_average should deal with it.

It's like added values in pandas with a channel index.

jona-sassenhagen, 2018-08-06:
Ok, renaming.

If channels occur in different orders for combine_evoked, it should internally reorder a copy of every dataset to have the same order as the first before averaging.