mne-tools/mne-python #5417: mne coreg gui, no isotrak, unhelpful error

jona-sassenhagen, 2018-08-06:
Trying to coreg my barebones EEG files, the Gui crashes with the rather unhelpful line:

```
...
isotrak = isotrak[0]
IndexError: list index out of range
```

I assume that means I don't have an Isotrak.

1. We should probably add a nicer error here.
2. What can I do instead :( I want to do horribly imprecise source localisation with my old EEG files. Can the suggested solution be included in the error?

larsoner, 2018-08-06:
We need to improve this pipeline, see #3893 #3877 #3699. But a good first start would be fixing whatever bug you are hitting here. The "isotrak" is probably meant to refer to `info['dig']`. So basically you need to populate this with your EEG electrode coordinates. If you use `DigMontage` instead of `Montage` I think this will happen automatically.

Fixing this the right way is #3893, feel free to look there if you think it makes sense / would help you here. I hope to do it (finally!) for 0.17.

jona-sassenhagen, 2018-08-06:
What I did now is simply read in the standard_1005 montage, raw.set_montage, and coregistered to that 😎 

I'd prefer not to help out there cause everything 3D is terribly confusing to me. (It's on some Star Trek movie IIRC)

mmagnuski, 2018-08-06:
I was creating a "fake" dig montage in your scenario, does it work for you? I've hit some errors on the road to successful coreg and also think it would be nice if the pipeline was easier.

jona-sassenhagen, 2018-08-06:
People who know stuff about source localisation might find this an aggressively bad idea, but couldn't we give a super streamlined pipeline for source localising default, non-digitised EEG data? In fact, we could provide precomputed inverse operators for standard_1005.

mmagnuski, 2018-08-06:
On the other hand digitizng chan pos with photogrammetry is easy and fun :)

larsoner, 2018-08-06:
> In fact, we could provide precomputed inverse operators for standard_1005.

For consistency and rigor I'd rather have people do the steps. Ideally it will just be a few easy lines once #3893 is done.

larsoner, 2019-08-21:
We have a tutorial for EEG now, and we've improved the code with a better error message
```
                error(None, "The selected FIFF file does not contain "
                      "digitizer information. Please select a different "
                      "file.", "Error Reading FIFF File")
```

closed 2019-08-21 by larsoner