mne-tools/mne-python #5420: BUG: BrainVision reads more events than it says it does.

sappelhoff, 2018-08-07:
According to the docstring, the BrainVision reader will only read "Stimulus" events, which are typically in the shape `r"(S\d{1,3})"`, so an S and then 1 to 3 digits. All other events will have to be specified using the `event_id` dictionary:
https://github.com/mne-tools/mne-python/blob/beda451422b4d8016643438ce7e998c76cbd2f59/mne/io/brainvision/brainvision.py#L233-L239

However, the BrainVision reader parses all events that end with a number, due to this regexp:
https://github.com/mne-tools/mne-python/blob/beda451422b4d8016643438ce7e998c76cbd2f59/mne/io/brainvision/brainvision.py#L325

This leads to issues when e.g., a "Comment" event is in the markers such as `Mk9=Comment,Discontinuit? [B] 4156 ms,136544,1,0` as in #5356. See [regexp101](https://regex101.com/r/VAqMnU/1) to play around with it.

That comment will then be automatically added to the events. Crucially, its trigger value will be the number that was found, in this case `4156`. That number originally referred to milliseconds. In this particular case the error is not too bad and will just lead to a nonsensical event value. However imagine that it was `254ms`: as BrainVision uses trigger values from 0 to 255, the 254 falls in the range of sensible event values and thus has the potency to be falsely added as a "true event" when it really is just a misread comment.

agramfort, 2018-08-07:
do you plan to send PR?

sappelhoff, 2018-08-07:
Yes, but first we need to discuss how to fix this issue. I could replace the present `r'[A-Za-z]*\s*?(\d+)'` with `r'S(\d{1,3})'` to truly match only the S### events as advertised by the docstring. However a lot of our tests are actually working with the buggy behavior. So it would involve fixing a lot of tests as well.

I'd be up for that but perhaps somebody has objections or a better idea.

larsoner, 2018-08-07:
How about a new kwarg `event_parsing='whatever' | 'full'` where `whatever` is some term that accurately describes what we currently do, and `full` is what you propose to change it to?

sappelhoff, 2018-08-08:
@larsoner thanks for the suggestion. However, what we currently do is really not sensible and I would like to change it, I am just looking for the best way to do it.

The arguably most common event in BrainVision files next to stimulus events marked as `S###` are response events marked as `R###`. These are also the most common events in our test files. I suggest to replace the current, faulty regexp in our code with [`[SR]([\s\d]{2}\d{1})`](https://regex101.com/r/UYbiJj/1), which will match any S### or R###.

Then I would adjust the docstrings to explicitly mention that.

What would remain is fixing the tests, because currently, we have an "optical" event in the test files with the marker `0  1`. The tests are tailored such that this event is read by default. But this event has a very idiosyncratic marker which we should not read by default (instead only when specified in `event_id`)

agramfort, 2018-08-08:
that's possible to change it if we consider a bug fix.

how does fieldtrip or eeglab or brainstorm handle?

if you tell me if makes mne behaves like them then I can live with the
bug fix option and hard API breakage.

sappelhoff, 2018-08-08:
@agramfort 

fieldtrip reads 
- stimuli (S###)
- responses (R###)
- new segments (we use that only to infer the recording measurement date)
- "time" (which I haven't heard of)

https://github.com/fieldtrip/fieldtrip/blob/d11daed366a426184c46bfb024a2cea7e04b1e19/fileio/private/read_brainvision_vmrk.m#L49-L77

... Brainstorm seems to read everything that is a marker by default:

https://github.com/brainstorm-tools/brainstorm3/blob/69bf5c064de3ac9864787d530d361c36546c7369/toolbox/io/in_events_brainamp.m#L32-L96

agramfort, 2018-08-08:
what do you suggest then?

sappelhoff, 2018-08-08:
I suggest we parse S### and R### (stimuli and responses) by default. everything else can be done with `event_id`

agramfort, 2018-08-08:
no objection.

PR #5426 (2018-08-09): Fix event parsing brainvision
fixes #5420 

TL;DR: We now by default parse events that are stimuli `S###` or responses `R###`, covering the bulk of events in BrainVision files. All other events have to be specified using `event_id`.

-----------------------
The changes to the tests were much more minimal than I expected. Basically I:
1. added `'0  1': 1` to the `event_id` so that the *optic event* in our test files gets parsed (before this bugfix it was wrongly parsed by default)
2. added the `event_id` argument to a `read_raw_brainvision` call where previously the `event_id` was apparently fo [... cut]
closed 2018-08-09 by agramfort