mne-tools/mne-python #5421: Clearify prerequisited for the installation and consequences of warnings.

tjochmann, 2018-08-07:
I tried to install MNE Python following the install instructions (`conda env create -f environment.yml`) on a new computer. It gave me a couple warnings:

- `Downloading https://files.pythonhosted.org/packages/36/8f/ab9555a37ab7c30560928c2eb0a9715f4492a11536f057aada019a319661/mayavi-4.6.1.tar.bz2 ...
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "C:\Users\tom\Anaconda3\envs\mne\lib\site-packages\setuptools\msvc.py", line 490, in _find_latest_available_vc_ver
        return self.find_available_vc_vers()[-1]
...
distutils.errors.DistutilsPlatformError: Microsoft Visual C++ 14.0 is required. Get it with "Microsoft Visual C++ Build Tools": http://landinghub.visualstudio.com/visual-cpp-build-tools`
This warning is particularly unfortunate, as the provided link to build tools is dead. I think we should add a functioning link and clearify, what part of MNE does not work without Mayavi. Installing the Build Tools for Visual Studio 2017 fixed this.
 - `twisted 18.7.0 requires PyHamcrest>=1.9.0, which is not installed.`
I guess this is just missing in the environment.yml.
 - `mkl-random 1.0.1 requires cython, which is not installed. 
mkl-fft 1.0.4 requires cython, which is not installed.`
Despite those warnings, a large part of MNE still works. I think someone who can oversee the consequences of those packages not being installed could add a few sentences to the install guide, giving recommendations when to make sure they are installed and when not to bother.

larsoner, 2018-08-07:
> This warning is particularly unfortunate, as the provided link to build tools is dead. I think we should add a functioning link and clearify, what part of MNE does not work without Mayavi. Installing the Build Tools for Visual Studio 2017 fixed this.

A compiler is only used for a single extension which is optional for Mayavi on latest `master`. In the meantime, doing this should work:
```
pip install https://api.github.com/repos/enthough/mayavi/zipball/master
```
We should probably update our `environment.yml` to use `master` or a pinned version until a 4.6.2 or later comes out.

The `mkl` warnings are probably not a big deal.

I have never seen the `PyHamcrest` problem but this sounds like a packaging problem with `twisted`.

larsoner, 2018-08-07:
(and yes updating the docs to say that the `Cython` warning and possibly the `PyHamcrest` warning is not critical is probably worthwhile)

schelian, 2018-09-21:
Hi @larsoner , I'm getting a 404 today (not a firewall issue).

larsoner, 2018-09-21:
There was a typo in the URL butb nowadays you should just be able to use the latest release version of Mayavi rather than git master

larsoner, 2018-09-21:
Closing as I think the instructions now mention warnings

closed 2018-09-21 by larsoner