mne-tools/mne-python #5425: make mne report BEM slider more efficient

jasmainak, 2018-08-08:
Curently, MNE report makes [3 copies](https://github.com/mne-tools/mne-python/blob/master/mne/report.py#L1780-L1785) of the MRI. This is an unnecessary waste of memory. Ideally, the slicing should be done on the fly -- as for example in @pbellec's [brainsprite library](https://github.com/SIMEXP/brainsprite.js/blob/master/assets/brainsprite.js)

agramfort, 2018-08-08:
it's because we make the PNGs for every slice offline. Do you expect a gain
in report size?

jasmainak, 2018-08-08:
yep, definitely. Right now report on sample data without BEM is 5.8 MB. And with BEM, it is 38.2 MB. If we did not have 3 copies, it would be 16.6 MB.

larsoner, 2018-08-08:
The three "copies" are actually different, i.e. one for each of three standard views. This seems like the most sensible default. I don't mind the file size personally, is it a limitation somewhere for you?

Maybe we can make it so render_bem can be a list of str of the subset of the three views to use, but I think that the default True should do what is done now.

What might be more useful is (losslessly) compressing the PNGs. There are a lot of black pixels in there. Sklearn does this somewhere IIRC using optipng or something

jasmainak, 2018-08-09:
It is not a copy _per se_, but you could construct the three views using javascript rather than python to save space. Anyway, it's not a limitation for me. It looked like an opportunity to optimize / sunday hack. But I won't complain if you want to close :-)

larsoner, 2018-08-09:
On-the fly slicing would be cool.

I also [tried PNG optimization](https://github.com/larsoner/mne-python/commit/65c6a5883eed2247147946f3a8612887238afafc) and it takes this:
```
import mne
data_path = mne.datasets.sample.data_path()
subjects_dir = data_path + '/subjects'
report = mne.Report()
report.add_bem_to_section('sample', subjects_dir=subjects_dir)
report.save('/Users/larsoner/Desktop/report.html', overwrite=True)
```
From 32.9 to 32.2 MB. So probably not worth it.

agramfort, 2018-08-09:
@jasmainak not that we only keep in the file the slices we display.

check that the original nii file is smaller what we already deliver.

Internal feeling: if it's not broken don't fix it...

larsoner, 2018-08-09:
Also I realized this is harder than just slicing the 3D MRI array, you also
need to do the contours by slicing the triangulated surfaces. So it would
require a real plotting library.

>

larsoner, 2018-08-09:
Closing as I actually think this would require using a proper plotting lib, which is probably a much larger change

closed 2018-08-09 by larsoner

jasmainak, 2018-08-10:
yeah, agreed. It seems broken. I [gave it a try](https://github.com/SIMEXP/brainsprite.js/issues/25) but couldn't get it to work out of the box.