mne-tools/mne-python #5427: np.nan in epochs.drop_log - `read_epochs` throws assert error

mmagnuski, 2018-08-08:
I got an epochs file saved in mne `16.2` from a collegue. I cannot read the file because the `drop_log` contains `[np.nan]` instead of `[]` where epochs were dropped. 
When reading the file with `mne.read_epochs` assertion error is thrown by `_check_consistency` (which was introduced to mne in #5193) - it does not take into account nans in the `drop_log`.
My collegue is able to read the file without problems.

Which one is a bug:
* saving file with `np.nan` in `drop_log`
* not taking into account `np.nan` in `drop_log` when reading
?

larsoner, 2018-08-08:
The drop log list should only contain strings, I don't know how nan could have gotten in there

mmagnuski, 2018-08-09:
Ok, thanks, I'll check how he gets those nans.

larsoner, 2018-08-09:
If they are inserting themselves using manual dropping, then it means 1) our check for the type of `reason` in [epochs.drop](https://martinos.org/mne/stable/generated/mne.Epochs.html#mne.Epochs.drop) should be improved and 2) we should add a check on writing that `drop_log` is `list of list of str` in case someone modifies `epochs.drop_log` themselves.

mmagnuski, 2018-08-09:
Ok, sorry, false alarm, my collegue was saving `drop_log` to csv with pandas and then loading it to reconstructed `Epochs` which gave rise to `np.nan` where originally he had `[]`.

larsoner, 2018-08-09:
Okay thanks for the update. I think we need to action then

larsoner, 2018-08-09:
No action I mean

closed 2018-08-09 by larsoner