mne-tools/mne-python #5432: meas_date documentation inconsistency

massich, 2018-08-10:
In `mne.Info`, `meas_date` is a list of two ints.
https://github.com/mne-tools/mne-python/blob/ff8ebca4a9d2497d4f8613c11b42300ccd3c9ba4/mne/io/meas_info.py#L149

however, based on `_handle_meas_date`, `raw.info['meas_date']` can have other formats.
https://github.com/mne-tools/mne-python/blob/ff8ebca4a9d2497d4f8613c11b42300ccd3c9ba4/mne/annotations.py#L272-L281

(Also, I find `_handle_meas_date` naming confusing, plus I don't understand why is a free function. IMHO it should be raw object method. i.e: `raw.get_meas_date_in_absolute_seconds` but I don't have enough arguments to do a proper assessment)

larsoner, 2018-08-10:
> however, based on _handle_meas_date, raw.info['meas_date'] can have other formats.

I think after I/O round trip it is always 2 ints. We probably need to improve our raw readers to make it more consistent. We could add a test to `_test_raw_reader` to ensure each class did it correctly. Want to take a stab at it?

> IMHO it should be raw object method

I guess you could make it an `Info` method (better than `Raw` because `meas_date` exists in `Info`, which can be in other objects) if you want. However I don't see too much to gain in this case, and it would complicate the `git blame` of the function/method, so I'm -0 on this.

agramfort, 2018-08-11:
yes we need more consistency in the raw data readers.

regarding the floating functions I prefer these when private as users don't
get exposed
to them with using tab completion. End users should not see these functions.

massich, 2018-08-11:
@agramfort I'm not sure I fully follow your thought here.

agramfort, 2018-08-12:
_handle_meas_date cannot be discovered with tab so easily. Whereas if I have a method in Annotations I just use tab in an instance of annotations and I start seeing a bunch of private functions I should not see. Clear?

larsoner, 2018-11-07:
I think this is fixed

closed 2018-11-07 by larsoner