mne-tools/mne-python #5441: BUG: BV reader now fails for certain modern files

jona-sassenhagen, 2018-08-13:
The most recent software by BrainProducts writes BrainVision files with integer codes. E.g.

```
Brain Vision Data Exchange Marker File, Version 1.0

[Common Infos]
Codepage=UTF-8
DataFile=BSN17_vw.eeg

[Marker Infos]
; Each entry: Mk<Marker number>=<Type>,<Description>,<Position in data points>,
; <Size in data points>, <Channel number (0 = marker is related to all channels)>
; Fields are delimited by commas, some fields might be omitted (empty).
; Commas in type or description text are coded as "\1".
Mk1=New Segment,,1,1,0,20160727190324526000
Mk2=Stimulus, 24,38759,1,0
Mk3=Stimulus,200,39169,1,0
Mk4=Stimulus,198,42426,1,0
Mk5=Stimulus,199,42843,1,0
Mk6=Stimulus, 99,45641,1,0
Mk7=Stimulus,201,46051,1,0
Mk8=Stimulus, 47,48690,1,0
Mk9=Stimulus,200,49100,1,0
```

As of #5426, Triggers such as ` 24` will currently not be parsed. Moreover, the warning string is somewhat deceptive, stating that "triggers such as [ 24, 200, 198 ..." will be dropped, but the whitespace matters here; e.g., only the last entry here would match: `{"24": 24, 24: 24, " 24": 24}`.

Given that Pycorder is a manufacturer-supplied and open, Python-based software, we should support such event codes by default.

@sappelhoff

h-kolkhorst, 2018-08-13:
I agree, the commit introduces regressions so that loading brainvision files fails after updating to the current master.

In addition to the Pycorder issue, the new regex does not match response markers, which defeats the whole purpose of the ``trig_shift_by_type`` parameter (or the old ``response_trig_shift``). The tests only run because the ``R255`` marker was added to ``event_id`` (https://github.com/mne-tools/mne-python/blob/master/mne/io/brainvision/tests/test_brainvision.py#L65) .

hekolk, 2018-08-13:
I agree, the commit introduces regressions so that loading brainvision files fails after updating to the current master.

In addition to the Pycorder issue, the new regex does not match response markers, which defeats the whole purpose of the ``trig_shift_by_type`` parameter (or the old ``response_trig_shift``). The tests only run because the ``R255`` marker was added to ``event_id`` (https://github.com/mne-tools/mne-python/blob/master/mne/io/brainvision/tests/test_brainvision.py#L65) .

sappelhoff, 2018-08-13:
@hekolk why does this defeat the purpose of `trig_shift_by_type`? If someone has response markers, they can add those to the `event_id` and then they will be shifted.

hekolk, 2018-08-13:
"defeats the purpose" was probably a bit too harsh, but with the "old" behavior, one only needed to specify a ``trig_shift_by_type``/``response_trig_shift`` and not specify every possible response marker. For example, in my code, I just shift the response or optical markers in a paradigm-agnostic way in my IO routines without having to know the exact marker definitions and occuring markers.

I'm not sure whether I have been using the method "incorrectly", but the following code worked fine, but breaks now:
```
    cur_raw = mne.io.read_raw_brainvision(vhdr_file, preload=True,
                                          trig_shift_by_type={
                                              'response': markers.RESPONSE_TRIG_SHIFT,
                                              'optical': markers.OPTICAL_TRIG_SHIFT
                                          })
```

hekolk, 2018-08-13:
More generally, 

1. the information in the (traditional) marker files of specifying both a type (``Response``)  and a prefixed description (``R123``) for the marker seems to be redundant and with Pycorder this redundancy seems to be removed in favor of the response type. This would suggest to just ignore the prefix.

2. with the old implementation, users had the choice to handle non-stimulus markers either (i) individually for each marker based on the description  and specify ``event_id`` or (ii) handle them by type and specify ``trig_shift_by_type``/``response_trig_shift``. Different users might favor different choices (personally I prefer handling markers by type). Maybe it is suboptimal from an API point-of-view to offer two ways to do the same thing, but I don't think it is an issue big enough to break the code of everybody who used the second variant.

PR #5442 (2018-08-16): [MRG] fix brainvision markers
fixes #5441 

- Now reading 'S###' as the format of BrainVision Recorder and '###' as the format of PyCorder.
- Adjusted docstrings and warning message
- In the warning message, the dropped markers are now printed as they are (i.e., no potentially crucial whitespace is being dropped)

To Do:
- [x] adding some tests asserting that we do read PyCorder style events
- [x] adding some tests asserting that we do NOT read events (e.g., comment events) ending with integers or similiar (just come up with some weird things that should be rejected)
- [x] discussion whether t [... cut]
closed 2018-08-16 by jona-sassenhagen