mne-tools/mne-python #5446: Epochs

hendriksantosa, 2018-08-16:
Hi, I am wondering whether MNE has reweighed version for epochs function rather than reject version using some threshold (mag and grad)?

larsoner, 2018-08-16:
I'm not sure what you mean. But our `reject` and `flat` just remove (effectively equivalent to a binary weight with weight 0). If you want some weighted average, you can compute one manually with `NumPy` and `epochs.get_data()`.

hendriksantosa, 2018-08-16:
I mean for example if the signal larger than some threshold. It will still keep that signal but down weight it in the evoked process. Right now, the function only remove the outliers using 'reject=reject' (as you mentioned with binary weight).

larsoner, 2018-09-23:
Agreed, I think we can let users import/create their own functions for this easily enough now

closed 2018-09-23 by larsoner