mne-tools/mne-python #5451: Hey, How about add some new features?

Stevewsw, 2018-08-17:
Hey, thanks for this brilliant repo;

I'm analyzing the EEG data for 'sleep stage scoring';

# Favor 

### How can I cut one channel of .edf file into 30s-epoch data group? 

AND

### How can I mark the annotations to each 30s-epoch, or labeling the 'W, N1, N2, N3, R' labels to every 30s-epoch data?

# Data structure

1. **.edf :**  each one obtains 10 channels info.

2. **.csv:**  

|No.|StartTime |Label|
|:---:|:---:|:---:|
|1	|21:34:57	|WK|
|2	|21:35:27	|WK|
|3	|21:35:57	|WK|
|4	|21:36:27	|WK|
|5	|21:36:57	|WK|
|6	|21:37:27	|WK|
|7	|21:37:57	|WK|
|...|...|...|

# Ideal result 
**30s-epoch(1x(30*Fs))**

----

**30s-epoch ： WK
... ： ..
30s-epoch  ： N2**

# End
Please do not hesitate to let me know if you have any questions.
It is greatly grateful if it would be responded;

agramfort, 2018-08-17:
what dataset are you analysing?

you need to understand how raw.times work and that raw.info['sfreq'] is used.
then you can just split the raw data you can access how expained online.

mmagnuski, 2018-12-19:
Closing due to lack of response, feel free to reopen.

closed 2018-12-19 by mmagnuski